mcu-tools/mcuboot #176: Maximum number of image sectors is 120 instead of 128

vanwinkeljan, 2017-12-27:
The current maximum number of sectors per image is hard coded at 120: 
https://github.com/runtimeco/mcuboot/blob/7cc3261e2d1df66b556a2ceadb76b5b89ea81c2e/boot/bootutil/src/bootutil_priv.h#L112

According to the the design document (docs/design.md) the maximum supported sectors per image is 128.

mbolivar, 2018-01-04:
Thanks for the report!

> Note that changing BOOT_MAX_IMG_SECTORS to 128 makes it possible to boot form image slots that are larger then 120 sectors.

Is this an issue for a target you are using? Lucky you, as it sounds like you have a lot of flash :).

If so, perhaps this should be made a configurable value (MCUBOOT_MAX_IMG_SECTORS to follow the usual convention for build-time config naming), controllable via the build system.

vanwinkeljan, 2018-01-04:
Yes it was an issue for an stm32l476rg device which has 1MB of flash with a sector size of 2K, so even with 128 sectors per image I have space left :)

Making it a config value is maybe the best option as it will also reduce the memory footprint.

mbolivar, 2018-01-05:
> Making it a config value is maybe the best option as it will also reduce the memory footprint.

@d3zd3z, @utzig, what do you think about that?

I guess we'd want a common convention for setting this up between the supported RTOS. Right now Zephyr takes configuration using CONF_XXX variables at cmake time, but it doesn't seem like mynewt does the same thing -- any ideas?

utzig, 2018-01-05:
Yes, I think making this build configurable seems like the best option. Zephyr has the `CONF_` macros and Mynewt has `BOOTUTIL_` macros (using newt) but both end up as `MCUBOOT_` defines.

carlescufi, 2018-04-10:
Fixed

closed 2018-04-10 by carlescufi