mcu-tools/mcuboot #182: image.py: Error on sign a firmware

miyatsu, 2018-01-16:
Use tag v1.1.0, when I sign a firmware using the root-*.pem key provide by this repo, got some error message.
mcuboot is at the root dir of zephyr, and run this script at dir zephyr/sample/http_client
```
#!/bin/bash

mcuboot_path="../../../mcuboot";
firmware_path="build/stm32f4_explo/zephyr";
firmware_version=1

output_file_name="${firmware_path}/http_v${firmware_version}_debug.bin";

./${mcuboot_path}/scripts/imgtool.py sign \
	--key ${mcuboot_path}/root-ec-p256.pem \
	--header-size 0x200 \
	--align 8 \
	--version ${firmware_version} \
	--included-header \
	${firmware_path}/zephyr.bin ${firmware_path}/http_v0.0.1_debug.bin
```

I got error message below:
```
Traceback (most recent call last):
  File "./../../../mcuboot/scripts/imgtool.py", line 145, in <module>
    args()
  File "./../../../mcuboot/scripts/imgtool.py", line 142, in args
    subcmds[args.subcmd](args)
  File "./../../../mcuboot/scripts/imgtool.py", line 84, in do_sign
    key = load_key(args) if args.key else None
  File "./../../../mcuboot/scripts/imgtool.py", line 62, in load_key
    key = keys.load(args.key)
  File "/home/${user_name}/project/zephyr/mcuboot/scripts/imgtool/keys/__init__.py", line 66, in load
    if pk.key_size != 256:
AttributeError: '_EllipticCurvePrivateKey' object has no attribute 'key_size'
```

utzig, 2018-01-16:
Hmmm, I can't reproduce the error here but seems related to dependencies. Please paste the output of `pip show cryptography` here.

miyatsu, 2018-01-16:
The output of ```pip show cryptography``` get nothing.
When I checkout to tag v1.0, and run the same .sh, every thing is fine.
I will try this on other machine.

utzig, 2018-01-16:
Recently `imgtool.py` has moved to depending only on `cryptography` instead of a bunch of different python crypto libraries. Strangely you don't get import errors, but try again after installing `cryptography` (`pip install cryptography` with or without sudo depending on your setup).

closed 2018-02-02 by miyatsu

miyatsu, 2018-02-06:
Fixed after #199 , commit id f242901a85980741edb4f25e8080b7c79df12714