mcu-tools/mcuboot #186: [Zephyr] Use Kconfig instead of CMake variables for configuration

SebastianBoe, 2018-01-25:
Hi,

I'm not sure what the background for this code is:

```

> ########################
> # Configuration choices.
> ########################
> 
> # Set CONF_SIGNATURE_TYPE to determine the signature type used.
> # Currently, it should be set to either RSA or ECDSA_P256.
> #
> # To choose RSA (this is the default):
> #
> #     cmake -DCONF_SIGNATURE_TYPE=RSA [...]
> #
> # To use ECDSA_P256:
> #
> #     cmake -DCONF_SIGNATURE_TYPE=ECDSA_P256 [...]
> if (NOT DEFINED CONF_SIGNATURE_TYPE)
>   set(CONF_SIGNATURE_TYPE RSA)
> endif()
> 
```

but as a general rule it is preferable to use Kconfig rather than CMake variables for configuration.

This github issue can be considered a feature request.

SebastianBoe, 2018-01-25:
Ah, I see ... you need to know the configuration values before boilerplate.cmake, so it is not straightforward to use Kconfig here ...

Hmm, I wonder how this can be resolved cleanly ...

SebastianBoe, 2018-01-25:
Actually, I suspect that it is possible to get rid of the code that is using these variables so this issue can be resolved cleanly ...

mbolivar, 2018-01-25:
> Actually, I suspect that it is possible to get rid of the code that is using these variables so this issue can be resolved cleanly ...

Any suggestions? These are a direct port from environment variables used by the the old Makefile, which predated the addition of the MCUBoot application-specific Kconfig options.

SebastianBoe, 2018-01-26:
Re-opening as it seems to be possible to do this. See #189

PR #272 (2018-04-25): Switch to kconfig
This series:

- starts with a cleanup of the Kconfig file
- adds new options to replicate functionality handled via CMake CONF_xxx macros in the Zephyr CMakeLists.txt
- mandates that every supported platform provide an mcuboot_config/mcuboot_config.h (only Mynewt was providing this previously)
- migrates each CONF_xxx over to its Kconfig equivalent by progressively extending Zephyr's mcuboot_config.h
- finishes up with a little extra cleanup

Fixes: #186 

Other than bootutil_log.h (which is itself a platform abstraction layer), this has the side effect of removing all platfo [... cut]
closed 2018-04-25 by utzig