mcu-tools/mcuboot #188: [Zephyr] Bootloader build failure with arduino_101

SebastianBoe, 2018-01-25:
Hi,

I am unable to build a Zephyr bootloader for the board arduino_101. I think my environment is sane because I am successful with frdm_k64f.

Tested with Zephyr version be0aa2ac2c53d7146e666d08756c31d4ffad2205

Steps to reproduce:

```
sebo@mach:~/mcuboot/boot/zephyr/b$ export BOARD=arduino_101
sebo@mach:~/mcuboot/boot/zephyr/b$ cmake ..
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3.4") 
-- Selected BOARD arduino_101
Zephyr version: 1.10.99
prj.conf:15: warning: attempt to assign the value "y" to the undefined symbol MPU_ALLOW_FLASH_WRITE
Parsing Kconfig tree in /home/sebo/mcuboot/boot/zephyr/../../Kconfig
Using /home/sebo/zephyr/boards/x86/arduino_101/arduino_101_defconfig as base
Merging prj.conf
warning: UART_QMSI_0_HW_FC was assigned the value "y" but got the value "n" -- check dependencies
warning: UART_QMSI_0_HW_FC was assigned the value "y" but got the value "n" -- check dependencies
-- Generating zephyr/include/generated/generated_dts_board.h
-- The C compiler identification is GNU 6.2.0
-- The CXX compiler identification is GNU 6.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/sebo/zephyr-sdk-0.9.2-rc5/sysroots/x86_64-pokysdk-linux/usr/bin/i586-zephyr-elfiamcu/i586-zephyr-elfiamcu-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
-- Performing Test check_nostartfiles
-- Performing Test check_nostartfiles - Success
-- Performing Test check_nodefaultlibs
-- Performing Test check_nodefaultlibs - Success
-- Performing Test check_nostdlib
-- Performing Test check_nostdlib - Success
-- Performing Test check_static
-- Performing Test check_static - Success
-- Performing Test check_no_pie
-- Performing Test check_no_pie - Success
-- Performing Test check_fno_asynchronous_unwind_tables_C
-- Performing Test check_fno_asynchronous_unwind_tables_C - Success
-- Performing Test check_fno_pie_C
-- Performing Test check_fno_pie_C - Success
-- Performing Test check_fno_pic_C
-- Performing Test check_fno_pic_C - Success
-- Performing Test check_fno_strict_overflow_C
-- Performing Test check_fno_strict_overflow_C - Success
-- Performing Test check_Wno_pointer_sign_C
-- Performing Test check_Wno_pointer_sign_C - Success
-- Performing Test check_Wno_unused_but_set_variable_C
-- Performing Test check_Wno_unused_but_set_variable_C - Success
-- Performing Test check_fno_reorder_functions_C
-- Performing Test check_fno_reorder_functions_C - Success
-- Performing Test check_fno_defer_pop_C
-- Performing Test check_fno_defer_pop_C - Success
-- Performing Test check_Werror_implicit_int_C
-- Performing Test check_Werror_implicit_int_C - Success
-- Performing Test check_Wl__X
-- Performing Test check_Wl__X - Success
-- Performing Test check_Wl__N
-- Performing Test check_Wl__N - Success
-- Performing Test check_Wl___gc_sections
-- Performing Test check_Wl___gc_sections - Success
-- Performing Test check_Wl___build_id_none
-- Performing Test check_Wl___build_id_none - Success
-- Performing Test check_ffunction_sections_C
-- Performing Test check_ffunction_sections_C - Success
-- Performing Test check_fdata_sections_C
-- Performing Test check_fdata_sections_C - Success
-- Performing Test check_mno_sse_C
-- Performing Test check_mno_sse_C - Success
-- Performing Test check_march_lakemont_C
[... 137 lines cut ...]
             .fa_size = FLASH_AREA_IMAGE_SCRATCH_SIZE,
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../flash_map.c: In function ‘flash_area_get_bounds’:
../flash_map.c:216:14: warning: assignment makes integer from pointer without a cast [-Wint-conversion]
         *off = FLASH_AREA_IMAGE_0_OFFSET;
              ^
../flash_map.c:217:14: warning: assignment makes integer from pointer without a cast [-Wint-conversion]
         *len = FLASH_AREA_IMAGE_0_SIZE;
              ^
../flash_map.c:220:14: warning: assignment makes integer from pointer without a cast [-Wint-conversion]
         *off = FLASH_AREA_IMAGE_1_OFFSET;
              ^
../flash_map.c:221:14: warning: assignment makes integer from pointer without a cast [-Wint-conversion]
         *len = FLASH_AREA_IMAGE_1_SIZE;
              ^
../flash_map.c:224:14: warning: assignment makes integer from pointer without a cast [-Wint-conversion]
         *off = FLASH_AREA_IMAGE_SCRATCH_OFFSET;
              ^
../flash_map.c:225:14: warning: assignment makes integer from pointer without a cast [-Wint-conversion]
         *len = FLASH_AREA_IMAGE_SCRATCH_SIZE;
              ^
[34/177] Building C object zephyr/CMakeFiles/zephyr.dir/arch/x86/soc/intel_quark/quark_se/eoi.c.obj
[35/177] Building C object zephyr/CMakeFiles/zephyr.dir/arch/x86/core/cache.c.obj
ninja: build stopped: subcommand failed.
```

mbolivar, 2018-01-26:
```
In file included from ../main.c:25:0:
../include/target.h:14:10: error: #include expects "FILENAME" or <FILENAME>
 #include MCUBOOT_TARGET_CONFIG
          ^~~~~~~~~~~~~~~~~~~~~
```

This is the problem. MCUBOOT_TARGET_CONFIG is not getting defined to "arduino_101.h" properly, even though boot/zephyr/targets/arduino_101.h exists. It looks like the build system sees it and is trying to make the definition, but there is some extra quoting:

```
ccache /home/sebo/zephyr-sdk-0.9.2-rc5/sysroots/x86_64-pokysdk-linux/usr/bin/i586-zephyr-elfiamcu/i586-zephyr-elfiamcu-gcc -DENABLE_EXTERNAL_ISR_HANDLING -DKERNEL -DMBEDTLS_CONFIG_FILE=\"config-boot.h\" -DMCUBOOT_TARGET_CONFIG="'\"arduino_101.h\"'" [...]  -c ../hal_flash.c
```

I've fixed the extra quoting here:

https://github.com/runtimeco/mcuboot/pull/190

But there are still build errors:

```
In file included from /home/mbolivar/src/mcuboot/boot/zephyr/include/target.h:14:0,
                 from /home/mbolivar/src/mcuboot/boot/zephyr/main.c:25:
/home/mbolivar/src/mcuboot/boot/zephyr/main.c: In function ‘main’:
/home/mbolivar/src/mcuboot/boot/zephyr/targets/arduino_101.h:30:28: error: ‘CONFIG_SOC_FLASH_QMSI_DEV_NAME’ undeclared (first use in this function)
 #define FLASH_DRIVER_NAME  CONFIG_SOC_FLASH_QMSI_DEV_NAME
                            ^
/home/mbolivar/src/mcuboot/boot/zephyr/main.c:103:44: note: in expansion of macro ‘FLASH_DRIVER_NAME’
     boot_flash_device = device_get_binding(FLASH_DRIVER_NAME);
                                            ^~~~~~~~~~~~~~~~~
/home/mbolivar/src/mcuboot/boot/zephyr/targets/arduino_101.h:30:28: note: each undeclared identifier is reported only once for each function it appears in
 #define FLASH_DRIVER_NAME  CONFIG_SOC_FLASH_QMSI_DEV_NAME
                            ^
/home/mbolivar/src/mcuboot/boot/zephyr/main.c:103:44: note: in expansion of macro ‘FLASH_DRIVER_NAME’
     boot_flash_device = device_get_binding(FLASH_DRIVER_NAME);
                                            ^~~~~~~~~~~~~~~~~
CMakeFiles/app.dir/build.make:302: recipe for target 'CMakeFiles/app.dir/main.c.obj' failed
make[2]: *** [CMakeFiles/app.dir/main.c.obj] Error 1
CMakeFiles/Makefile2:227: recipe for target 'CMakeFiles/app.dir/all' failed
```

It looks like CONFIG_FLASH=y, but the QMSI flash driver is not being selected. I'd say this is a problem to fix in the board's Kconfig.defconfig.

aditihilbert, 2018-03-13:
@andrewboie do you have any ideas for resolving this issue?

PR #247 (2018-03-13): zephyr: Fix overlay dir for Zephyr build
The Zephyr CMakeLists.txt uses its own logic to have board-specific
overlays, named boards/${BOARD}.prj.  However, the only overlay file we
have, arduino_101.conf, isn't in this boards directory, and this target
doesn't build.

Fix this by moving the target into the right directory.  Fixes #188.
commit d9d28a78 (2018-03-13): zephyr: Fix overlay dir for Zephyr build

The Zephyr CMakeLists.txt uses its own logic to have board-specific
overlays, named boards/${BOARD}.prj.  However, the only overlay file we
have, arduino_101.conf, isn't in this boards directory, and this target
doesn't build.

Fix this by moving the target into the right directory.  Fixes #188.
closed 2018-03-13 by d3zd3z