mcu-tools/mcuboot #200: Add more "how to" doc for zephyr targets

carlescufi, 2018-01-30:
Looks like the description is too dry. It needs to be more guide.

I observe that:
it is not clear to user how-to compile proper image-application and how to create the image for upgarde
https://github.com/runtimeco/mcuboot/blob/master/README-zephyr.rst#building-applications-for-the-bootloader
Prefered here is examle of what should be added to the project cmake file and how a-board-overlay.dts should looks like. For instance we can redact this somehow:

```
Put into the project's top cmake the path to the overlay file: 

set(DTC_OVERLAY_FILE "${CMAKE_CURRENT_SOURCE_DIR}/board-dts.overlay") before including boilerplate.cmake, File should looks for instance like this:
set(CONF_FILE "prj.conf")

set(DTC_OVERLAY_FILE "${CMAKE_CURRENT_SOURCE_DIR}/dts.overlay")

include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)
project(NONE)

target_sources(app PRIVATE src/main.c)
 

add the file <a-board.overlay> with payload:

/ {
	chosen {
		zephyr,code-partition = &slot0_partition;
	};
};
```

for: https://github.com/runtimeco/mcuboot/blob/master/README-zephyr.rst#signing-the-application need to exaplane --pad imagetool option user case (manually triggered test image swap).

utzig, 2018-01-30:
@nvlsianpu FYI

mbolivar, 2018-01-30:
Fixed link:

https://github.com/runtimeco/mcuboot/blob/master/docs/readme-zephyr.md#building-applications-for-the-bootloader

> Prefered here is examle of what should be added to the project cmake file and how a-board-overlay.dts should looks like.

You're saying a method of starting with an existing sample and "MCUboot-izing" it, rather than showing a working example people can extend.

Since there are several pull requests just getting merged around this in both repositories, perhaps we can settle on the solution and then handle this issue?

utzig, 2018-01-30:
@mbolivar This is a copy/paste from a ticket in Jira... we are testing moving stuff from there to here! (https://runtimeco.atlassian.net/browse/MCUB-95)

carlescufi, 2018-01-30:
@mbolivar yes, sorry this was just a transition from Jira to GitHub issues, I didn't really create the issue now.

mbolivar, 2018-01-30:
@utzig @carlescufi got it, thanks!

nvlsianpu, 2018-01-31:
There is idea that the guide should be added to zephyr-project doc and be pointed form mcuboot doc. This was oryginalny @utzig suggestion on that. TEST :D!

carlescufi, 2018-03-27:
I am working on this as part of the MCUmgr documentation work I am doing

carlescufi, 2018-04-03:
Will fix this in time for 1.2

carlescufi, 2018-04-05:
This is now fixed by previous MCUboot PRs and also by https://github.com/zephyrproject-rtos/zephyr/pull/6917

closed 2018-04-05 by carlescufi