mcu-tools/mcuboot #202: Clean up Zephyr flash driver name after 1.11

mbolivar, 2018-01-31:
The MCUboot define FLASH_DRIVER_NAME needs to be globally renamed to FLASH_DEV_NAME after Zephyr 1.11.

Further reading: https://github.com/runtimeco/mcuboot/pull/201

d3zd3z, 2018-02-01:
Should we keep compatibility with more than just a single version of Zephyr? For example, if we wanted to be compatible with two versions, this would get cleaned up, maybe 1.12?

Or, are we willing to just always require the latest release of Zephyr for any given version of MCUboot?

mbolivar, 2018-02-02:
My preference would be to stay on mainline at least until Zephyr LTS 1.12 is released. We can make efforts to maintain compatibility with 1.12 when that's out. Zephyr is just moving too quickly right now to try to stay in sync with multiple versions.

PR #267 (2018-04-10): zephyr: cleanup flash driver name
The flash driver name was changed upstream during the v1.11
development cycle. Remove a shim which kept the old name around in the
Zephyr port now that v1.11 is out.

Fixes: #202
closed 2018-04-10 by utzig