mcu-tools/mcuboot #203: Allow public keys to be specified at build time

d3zd3z, 2018-02-05:
Instead of baking the specific public keys directly into the build, allow cmake variables (for Zephyr) to point to a file containing the keys to be built in.

mbolivar, 2018-02-06:
Can we use Kconfig for this if possible? Be good to start the move over to it

utzig, 2018-03-13:
What's the idea here? User supplies an input `.pem`/`.der` file in the configuration and *the script* generates the c files?

mbolivar, 2018-03-13:
What would be nice to see from my perspective is:

User supplies a pubkey file (or files) in some format (whatever formats imgtool generates, let's say). The build system ensures the raw keys are present in the key array mcuboot uses to verify signatures. Not sure what you mean by "the script".

d3zd3z, 2018-03-15:
From a "mild" deployment point of view, what @mbolivar suggests is probably best. Some way, variable or kconfig tells the build system where the public key(s) come from. These are considered source files. The C files of the public key is built by the build system (and would be removed by `clean`).

In a more serious type of development environment, the developers probably won't have access to the private keys. This is why I made it possible to separate the two. That way this solution could still be used to embed the public key into the image, and the private key would probably live inside of a hardware security module (HSM), which would be more controlled. In this case, developers might have to give the images to someone else to have them be signed.

utzig, 2018-04-03:
This would be cool to have on Mynewt side as well... I would suggest something like:

```
target.app: "apps/blinky"
target.bsp: "hw/bsp/stm32f3discovery"
target.build_profile: "debug"
target.public_key: f3disco_key.pem
```

`newt` would then generate the required `keys`? package with the expected `bootutil_keys` info. What do you think? /cc @mkiiskila @ccollins476ad @sterlinghughes

mbolivar, 2018-04-05:
> In a more serious type of development environment, the developers probably won't have access to the private keys. This is why I made it possible to separate the two. That way this solution could [...]

I don't follow why that affects the mcuboot build.

All you need to build mcuboot is the public key.

ccollins476ad, 2018-04-05:
@utzig - I think that is a great idea.  Embedding the public key in the boot loader has always been a headache.  Specifying the public key filename in the target seems like the right way to make this more convenient.

PR #275 (2018-06-06): Autogen keys
Add support for describing the key file in the configuration.

Feature complete but might be in need of some cleanups. Suggestions are welcomed!

This fixes #203
closed 2018-06-06 by utzig