mcu-tools/mcuboot #204: Define communication between bootloader and booted image

d3zd3z, 2018-02-05:
In order for the application to know how to perform an upgrade, it needs to know certain information from the bootloader, at a minimum, a version number. It would be better, however, for this information to be closer to the information needed to convey (upgrade type, slot information, etc).

Implement a mechanism to communicate this information between the bootloader and the running app.

d3zd3z, 2018-02-05:
Copied from [MCUB-17](https://runtimeco.atlassian.net/browse/MCUB-17).

nvlsianpu, 2018-02-13:
[Zephyr: Inter-applications flash_map exchange mechanism](https://github.com/zephyrproject-rtos/zephyr/issues/6152)

d3zd3z, 2019-11-27:
I've requested [Zephyr: Inter-applications flash_map exchange mechanism](https://github.com/zephyrproject-rtos/zephyr/issues/6152) be closed, and I'm going to close this as well. If someone finds a need for this, feel free to re-open, but with Zephyr, we seem to do OK by conveying this information through device tree, and the current mechanism.

closed 2019-11-27 by d3zd3z