mcu-tools/mcuboot #205: Implement dual-slot (non swap) image upgrade

d3zd3z, 2018-02-05:
Implement a mechanism that allows two different images to be deployed, one that runs in slot 0 and one that runs in slot 1. MCUboot's responsibility will then be to track which one to run, not to handle the complexity of swapping them.

leodesigner, 2018-05-28:
+1 
The similar way as it done in ESP8266 OTA update. It would be very nice feature during development to preserve flash write count. In this case we must prepare two firmwares linked to the slot0 and slot1. And later during the OTA the right image will be chosen for non used currently slot.

tamasban, 2018-05-28:
We have an implementation, it will be published next week on Lianro gerrit TF-M repo. I will post a link for the review here.

leodesigner, 2018-05-28:
Great !
It is really good option. I hope it will be easy to add to the Zephyr cmake linking option for slot1.

uday-agarwal, 2018-12-20:
@tamasban can you please post the link if you have it?

tamasban, 2018-12-20:
@uday-agarwal
Hi, 
We have a forked MCUBoot repo in TF-M. Currently the non-swapping update only up-streamed to there:
https://git.trustedfirmware.org/trusted-firmware-m.git/tree/bl2/ext/mcuboot

The two repo is currently diverged but will be synced.

Tamas

uday-agarwal, 2018-12-20:
Thanks a lot Tamas.

d3zd3z, 2019-12-02:
I believe this is called "noswap" in the TF-M code. Assigning to Tamas for tracking.

nvlsianpu, 2020-02-13:
@hakonfam I received some interest on this feature request - what do you think, do you see any obstacles?
Implementation shouldn't be so complex I believe - so I would provide  a POC.

hakonfam, 2020-02-13:
> @hakonfam I received some interest on this feature request - what do you think, do you see any obstacles?
> Implementation shouldn't be so complex I believe - so I would provide a POC.

We already have this feature downstream in NCS. It will be generalized in what is called "jerryboot" to which there is an open PR.

Laczen, 2020-02-21:
@hakonfam, how do you handle encryption in such a setup?

hakonfam, 2020-02-21:
> @hakonfam, how do you handle encryption in such a setup?

Currently it is not supported.
But the the way I think it should work is that it would be up to the component storing the update to perform the decryption.
The decryption would be performed using an API exposed from the bootloader.
This allows the application to read the cleartext update, which is OK imo, since its an updated version of itself.

utzig, 2020-02-21:
> But the the way I think it should work is that it would be up to the component storing the update to perform the decryption.

I am not sure I understand your suggestion, but can't decrypt in place, the only way would be to load and decrypt to RAM and run from RAM.

Laczen, 2020-02-21:
I don't think it is a good idea to use the component storing the update to do the decryption, then all security is transferred to this component.

There is a way (which is implemented in zb-8) to avoid this. zb-8 allows for "in-place" decryption. In this case the upgrade image is placed one flash sector higher than the run image. The bootloader detects that there is a new image (a crc-32 check fails) and decrypts while "dropping" the image one flash sector down. In zb-8 this is used to provide support for devices with small flash and the bootloader has a (unsecure) upload method. In case there is sufficient flash the same method could be used for two images, where the present running image would place the new image in the appropiate slot. All you loose is two sectors of flash, but encryption is maintained.

hakonfam, 2020-02-21:
> > But the the way I think it should work is that it would be up to the component storing the update to perform the decryption.
> 
> I am not sure I understand your suggestion, but can't decrypt in place, the only way would be to load and decrypt to RAM and run from RAM.

It would not be in place, the alternate slot is running. So code running from slot 0 would decrypt into slot 1.

Laczen, 2020-02-21:
> It would not be in place, the alternate slot is running. So code running from slot 0 would decrypt into slot 1.

Ooh, you would add the decryption to the application, in that case there is no need for a bootloader is there?

hakonfam, 2020-02-21:
I feel like we are polluting this issue, lets take this on slack :)

closed 2021-06-04 by github-actions[bot]

nvlsianpu, 2021-06-04:
This feature request was fulfilled by #739.