mcu-tools/mcuboot #218: scratch area size might be used as flash-sector-size (zephyr)

aditihilbert, 2018-02-05:
For zephyr targets the scratch area size is used as the flash sector size in case API for retrieving the layout of flash memory pages.is disabled (CONFIG_FLASH_PAGE_LAYOUT == n). 

This causes in flash area size verification error.
This bug is related at last to nrf52xx targets 

-----------
Andrzej Puzdrowski added a comment - 23/Nov/17 6:12 AM

Values form directory https://github.com/runtimeco/mcuboot/tree/3d9e3eb974812b0a02c3185ab15c2f7ce9a99feb/boot/zephyr/targets
should be used instead of {quote}#define FLASH_AREA_IMAGE_SECTOR_SIZE FLASH_AREA_IMAGE_SCRATCH_SIZE{quote}

mbolivar, 2018-02-05:
Mcuboot on zephyr really needs CONFIG_FLASH_PAGE_LAYOUT == y.

Building without it is always going to be a hack.

Is there a reason not to close this as not a bug?

d3zd3z, 2019-12-02:
No response, closing as not a bug.

closed 2019-12-02 by d3zd3z