mcu-tools/mcuboot #225: Support multiple flash devices (slot 0 internal, slot 1+scratch external)

aditihilbert, 2018-02-05:
There are some concerns that need to be addressed to support a configuration where slot 0 is on an internal flash device, and slot 1 and scratch are on a larger external flash. This allows most of the internal flash to be used for code. 

* Although there is some support in the code, it hasn't been tested, so will need simulator support to test this. 
* The Zephyr port is hardcoded for a single flash device. 
* How to handle the write alignment being different between internal and external flash devices. 
* Addressing issues discovered by the above.

nvlsianpu, 2018-08-08:
subtask: https://github.com/runtimeco/mcuboot/issues/326

utzig, 2019-09-13:
> * Although there is some support in the code, it hasn't been tested, so will need simulator support to test this.
> * How to handle the write alignment being different between internal and external flash devices.

Both those features have been implemented already. I am not sure if this is already working on Zephyr, but on Mynewt is totally works.

d3zd3z, 2019-10-23:
I think we can close this.

utzig, 2019-10-23:
Yes, this works in Mynewt and Zephyr.

d3zd3z, 2019-11-04:
Although the description above mentions putting scratch on the external device, which is definitely not what you want for encrypted images, and probably not what you want for even regular upgrades (as it provides an area for attack when interrupting an update.

Otherwise, though, I do think this is no longer relevant.

closed 2019-12-02 by d3zd3z