mcu-tools/mcuboot #226: Detect oversized app image

aditihilbert, 2018-02-05:
Right now, if an application overlaps the image trailer, it just causes failures. Ideally, we should detect this, but there are difficulties since this information depends on the flash alignment, which isn't readily available in the host tools.

First question - can we detect this via Newt?

ccollins476ad, 2018-02-05:
Newt is able to detect this condition via the `MCU_FLASH_MIN_WRITE_SIZE` setting.  This setting indicates the MCU's minimum flash write size.  Newt uses this setting to calculate the size of a hypothetical image trailer, and therefore the maximum allowed size of the image data.

SebastianBoe, 2018-02-20:
If the image trailer was built into the image when the application was first linked/built one could detect this (linker error) by having a memory region that limited the space used.

(This approach solves other problems as well)

mbolivar, 2018-02-20:
> If the image trailer was built into the image when the application was first linked/built one could detect this (linker error) by having a memory region that limited the space used.
> 

Interesting idea!

> (This approach solves other problems as well)

But it causes additional ones: now all application binary images are as large as the slot itself (with padding bytes in between), which wastes bandwidth to FOTA, etc.

A linker-based approach, I think, should instead force the linker script to leave room for the trailer in the memory entry for ROM, but not actually include it.

What do you think?

SebastianBoe, 2018-02-21:
> A linker-based approach, I think, should instead force the linker script to leave room for the trailer in the memory entry for ROM, but not actually include it.

That is better than today's solution at least, so if not my proposal then yes indeed.

> But it causes additional ones: now all application binary images are as large as the slot itself (with padding bytes in between), which wastes bandwidth to FOTA, etc.

? I don't think so, my claim is that it is possible to get the same hex as you do when you post-process with imgtool. I can prove this by example later, but it is interesting for me to hear now if there are any other problems with this approach.

mbolivar, 2018-02-21:
> I don't think so, my claim is that it is possible to get the same hex as you do when you post-process with imgtool.

Operative word being "hex", right? When I said "binary images" I meant "bin"

SebastianBoe, 2018-02-21:
Ah, right, but I assume one would never need to FOTA a binary image? I am not familiar with this domain.

mbolivar, 2018-02-21:
We only ever FOTA binary images (over IP), actually -- the Zephyr applications don't have .hex parser, so they just receive the signed blob and write it directly to slot 1, then reset.

utzig, 2020-01-15:
This is very hard to do in the bootloader itself (impossible maybe?) because checking would have to be done before beginning a swap. The problem is that once a swap is under way, and the device is reset, the checking will possibly fail on next boot because one cannot determine the size of the image. It would be possible to read `swap_size` from the trailer maybe, but it's hard to know if it's correct. For the swap-move implementation this was straight-forward because only sector indexes are required, since trailer and image data can never exist on the same sectors.

So part of fixing this is adding creation checks, like done in PR #636 for `imgtool`. A similar change needs to be added to `newt`. Another thing to check is if there is overlap of trailer in the scratch area with image data, in those cases where an image is big enough to require the trailer to be created on scratch (this would also be the case in some STM32 with large sectors).

PR #637 (2020-01-16): Fix imgtool checking of oversized app
This partly fixes #226 (the image generation side). Previous PR (#636) was wrong!
closed 2020-01-16 by utzig