mcu-tools/mcuboot #229: readme-zephyr.md has a broken link

SebastianBoe, 2018-02-12:
The Zephyr docs have a broken link when rendered through GitHub. It looks like this:

Please see the [design document]({% link design.md %}) for documentation on the design and operation of the bootloader itself. This functionality should be the same on all supported RTOSs.

carlescufi, 2018-02-12:
This is unfortunate, it's because the same .md file is used for the website, which uses that templating:

https://mcuboot.com/readme-zephyr.html

carlescufi, 2018-02-12:
@utzig any solution to this?

SebastianBoe, 2018-02-12:
Also, on the frontpage, this link is broken:

imgtool: A tool to securely sign firmware images for booting by mcuboot.

utzig, 2018-02-12:
I removed the `link` in the docs with #231. From what I read about gh-pages it would allow access both on while browsing and mcuboot.com this way. But it might not work and need a revert later...

SebastianBoe, 2018-02-13:
@utzig : Great. The readme-zephyr link is fixed. But the frontpage "imgtool" link is still broken.

closed 2018-02-13 by SebastianBoe

utzig, 2018-02-13:
@SebastianBoe Both links work for me (mcuboot.com and gh/runtime/mcuboot). How are you accessing it?

SebastianBoe, 2018-02-13:
@utzig : From here: https://github.com/runtimeco/mcuboot

I click on "imgtool".

This redirects me to

https://github.com/runtimeco/mcuboot/tree/master/imgtool

Which GH gives a 404 for.

utzig, 2018-02-13:
Thanks, missed the ones in `README.md`. PR #234 should fix those.