mcu-tools/mcuboot #230: [Zephyr] # CONFIG_IS_BOOTLOADER is not set

SebastianBoe, 2018-02-12:
Hi,

when following the instructions for building a Zephyr bootloader I was surprised to see that in the .config:

`# CONFIG_IS_BOOTLOADER is not set`

Is this a bug?

SebastianBoe, 2018-02-12:
Steps to reproduce:

Build ~/mcuboot/boot/zephyr with BOARD nrf52_pca10040.

grep IS_BOOT ~/mcuboot/boot/zephyr/build/zephyr/.config

carlescufi, 2018-02-12:
No, this is not a bug. `CONFIG_IS_BOOTLOADER` is essentially a hack. We do not want cross-dependencies between Zephyr and MCUboot. Instead we only want MCUboot to be able to configure Zephyr to meet its requirements. If anything what we need to do is to get rid or rename `CONFIG_IS_BOOTLOADER`.

carlescufi, 2018-02-12:
@utzig we can close this.

closed 2018-02-12 by utzig

SebastianBoe, 2018-02-13:
okay ...

then we definitely need to improve the documentation here:

```
config IS_BOOTLOADER
	bool "Act as a bootloader"
	default n
	depends on XIP
	depends on ARM
	help
	  This option indicates that Zephyr will act as a bootloader to execute
	  a separate Zephyr image payload.

```

When is IS_BOOTLOADER supposed to be enabled/disabled?

mbolivar, 2018-02-13:
CONFIG_IS_BOOTLOADER's purpose is to set aside an area of SRAM for use by Zephyr. Perhaps that is useful on some bootloaders if they need to keep their state after passing control to the application. But I think this is not necessary on mcuboot, which does not need any state preserved after passing control to Zephyr. (We discussed passing some mcuboot state to Zephyr a while ago on the developer list, but this is not a supported feature now.)

SebastianBoe, 2018-02-13:
Right, so maybe,

CONFIG_RESERVE_RAM_FOR_BOOTLOADER

or something. Because obviously it is possible to be a bootloader, and not need RAM reserved.
I would suggest that this issue is re-opened with a different title.

mbolivar, 2018-02-13:
That's a Zephyr issue, not an MCUboot one

SebastianBoe, 2018-02-13:
My mistake.