mcu-tools/mcuboot #233: Downgrade prevention

nvlsianpu, 2018-02-13:
Optional feature:
Bootloader should not accept any firmware which has version lower than current one. It may be that older firmware has known vulnerability and attack wants to exploit it.

mkiiskila, 2018-02-13:
Users will want to backwards in versions. This is especially during testing, but also in the field.

utzig, 2018-02-13:
@mkiiskila I believe this was discussed in a meeting before, and the idea was to enable OPTIONAL support for this feature. This doesn't make sense for development, but could be wanted in the field. Also kinda in the same veins, @d3zd3z suggested using 'overwrite-only' mode when no fallback to previous version should ever be necessary.

SebastianBoe, 2018-02-13:
FYI: HomeKit does not allow downgrading FW.

mbolivar, 2018-02-13:
This is a duplicate of #224

closed 2018-02-13 by mbolivar