mcu-tools/mcuboot #240: Creating an unsigned image with imgtool requires using the "sign" subcommand

SebastianBoe, 2018-02-20:
Creating an unsigned image with imgtool requires using the "sign" subcommand. Either a new subcmd needs to be added, or the "sign" subcommand needs to renamed to "create".

utzig, 2018-02-20:
Using `create` seems like a good idea!

d3zd3z, 2018-03-15:
When you "sign" without using a key, it is still putting a sha hash at the end. I'm not sure if `create` really captures this either, but I'm not sure what the best word would be.

If we did add another command, it should probably just be an alias, since sign does really make sense for the cases where we sign the image.

What do you think of #251 as a solution to this?

SebastianBoe, 2018-03-20:
"What do you think of #251 as a solution to this?"

Review feedback given in #251.

aditihilbert, 2018-03-20:
how about a flag like -n (no sign, hash only)?

d3zd3z, 2018-04-05:
I don't think it would be helpful to have a flag to indicate no signature. If you give a key, it will be signed, if you don't give a key, it won't be signed. We just have to make the description clear enough. Any reviews on #251 would be appreciated.

PR #251 (2018-04-10): imgtool: Add 'create' alias for 'sign'
The imgtool's `sign` command also works without a key, but it doesn't
sign, only appends a hash.  Add a `create` alias to this command so that
this usage makes more sense.

Fixes #240
commit 5a18102c (2018-04-10): imgtool: Add 'create' alias for 'sign'

The imgtool's `sign` command also works without a key, but it doesn't
sign, only appends a hash.  Add a `create` alias to this command so that
this usage makes more sense.

Fixes #240
closed 2018-04-10 by d3zd3z