mcu-tools/mcuboot #253: samples/zephyr make error

Jeepgoing, 2018-03-21:
hello,when i make zephyr hello-world use "make hello1",an error appeared,who can help me:

Traceback (most recent call last):
  File "../../scripts/imgtool.py", line 145, in <module>
    args()
  File "../../scripts/imgtool.py", line 142, in args
    subcmds[args.subcmd](args)
  File "../../scripts/imgtool.py", line 85, in do_sign
    img.sign(key)
  File "/home/liang/work/shared/mcuboot-1.1.0/scripts/imgtool/image.py", line 128, in sign
    sig = key.sign(bytes(self.payload))
  File "/home/liang/work/shared/mcuboot-1.1.0/scripts/imgtool/keys/rsa.py", line 92, in sign
    return self.key.sign(
AttributeError: '_RSAPrivateKey' object has no attribute 'sign'

Jeepgoing, 2018-03-21:
i have installed scripts/requirements.txt with sudo and without sudo

Jeepgoing, 2018-03-21:
i have uninstall cryptography,and reinstall cryptography,this problem was solved.
i think when i first install cryptography,maybe i used sudo.

closed 2018-03-21 by Jeepgoing

aspirine92, 2018-07-31:
Actually cryptography package was not installed.

```
pip3 install cryptography
```

No "sudo" required.

d3zd3z, 2018-07-31:
Generally, should be

    pip3 install -u cryptography

Depending on your distribution. In general, you should *never* use sudo on a pip command.