mcu-tools/mcuboot #269: Unable to sign image using imgtool.py

yatishmishra, 2018-04-12:
I compiled the sample hello world and tried signing it with the following command
../../../../scripts/imgtool.py sign --key ../../../../root-rsa-2048.pem --header-size 0x200 --align 8 --version 1.2 --included-header ./zephyr/zephyr.bin signed.bin

I get following error

Traceback (most recent call last):
  File "../../../../scripts/imgtool.py", line 148, in <module>
    args()
  File "../../../../scripts/imgtool.py", line 145, in args
    subcmds[args.subcmd](args)
  File "../../../../scripts/imgtool.py", line 85, in do_sign
    img.sign(key)
  File "/home/yatish/Desktop/zephyr-zephyr-v1.11.0/samples/mcuboot-master/scripts/imgtool/image.py", line 136, in sign
    sig = key.sign(bytes(self.payload))
  File "/home/yatish/Desktop/zephyr-zephyr-v1.11.0/samples/mcuboot-master/scripts/imgtool/keys/rsa.py", line 92, in sign
    return self.key.sign(
AttributeError: '_RSAPrivateKey' object has no attribute 'sign'

utzig, 2018-04-12:
Have you installed the requirements based on `scripts/requirements.txt`? Please attach the output of `pip list`.

yatishmishra, 2018-04-12:
Thanks for reply. Here is the list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
adium-theme-ubuntu (0.3.4)
asn1crypto (0.24.0)
btproxy (0.1)
cffi (1.11.5)
cryptography (2.2.2)
cycler (0.9.0)
dtrx (6.6)
enum34 (1.1.6)
idna (2.6)
intelhex (2.2.1)
ipaddress (1.0.19)
matplotlib (1.5.1)
numpy (1.11.0)
paho-mqtt (1.3.0)
Pillow (3.1.2)
pip (9.0.1)
PyBluez (0.22)
pycparser (2.18)
pyparsing (2.0.3)
Pyste (0.9.10)
python-dateutil (2.4.2)
pytz (2014.10)
setuptools (36.2.7)
six (1.10.0)
unity-lens-photos (1.0)
wheel (0.29.0)
You are using pip version 9.0.1, however version 9.0.3 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

yatishmishra, 2018-04-12:
I updated PIP as well but see the same error.

utzig, 2018-04-12:
Well, `_RSAPrivateKey` clearly has a `sign` method:

https://github.com/pyca/cryptography/blob/master/src/cryptography/hazmat/backends/openssl/rsa.py#L411

I am really not sure what is going on...

@d3zd3z Have you seen this error before?

utzig, 2018-04-12:
Btw, which `Python` version are you using?

yatishmishra, 2018-04-12:
Python 2.7.12

utzig, 2018-04-12:
The first line of `imgtool.py` includes `#! /usr/bin/env python3`... so you probably need to run: `python3 --version`

utzig, 2018-04-12:
ah, you probably also need to install your dependencies with `pip3`, that might be the issue...

yatishmishra, 2018-04-12:
I uninstalled cryptography and re installed. That made it work. Thanks for all your support :)

closed 2018-04-12 by yatishmishra