mcu-tools/mcuboot #271: Why MCUBOOT_SERIAL restores in area0 directly ?

b-allibert, 2018-04-20:
Hi team,
I have a quick question : i see that MCUBOOT_SERIAL feature seems to restore an image into area0 (i.e. the one used for execution, right ?). 
But i'm wondering why this feature does not flash area1 and resets ? This would allow using the usual security mcuboot features for the next boot (i.e. authenticate and restore from area1 to area0 once authentication is OK)... but i guess i'm probably missing something here.
Thanks for your answer
Bastien

uday-agarwal, 2018-12-20:
Hi there,

I am facing the same problem. Seems to me that an image uploaded over serial is written directly to slot 0, and slot 1 is never used. Also, I can't seem to find support for image erase, test, and confirm commands. Please advise as to whether this is correct, and if there's any planned support in future. Right now this is limiting me to use a single image only, and I'm not able to store two images in the target. My target device is nRF52 running Zephyr with mcuboot.

Best Regards
Uday

utzig, 2018-12-20:
This functionality was intended to initialize a new equipment, not to install an upgrade image. Installing an upgrade image to slot1 is the job of the running firmware in slot0. You should check Zephyr's `mcumgr` support.

Also, the bootloader wouldn't really know what to do if serial bootloader downloaded an image to slot1 without an image already installed on slot0 because it needs some metadata (like image header) to decide what to do.

Regarding security features, just enable `CONFIG_BOOT_VALIDATE_SLOT0` option, so that it never runs the image in slot0 without validation.

uday-agarwal, 2018-12-20:
Got it. Thanks for the quick response! I'll look more closely at Zephyr support and come back for any followup questions. Thanks again.

d3zd3z, 2019-12-02:
This seems to be answered.

closed 2019-12-02 by d3zd3z