mcu-tools/mcuboot #277: mcuboot zephyr

Jeepgoing, 2018-05-03:
hi all:
i have used mcuboot for zephyr os and i find a problem,mcuboot must config 3 slots,slot0, solt1 and scratch area, my flash size is 512k and my slot is more than 128k, so i can not use 3 slots, only slot0 and slot1, but mcuboot has a limition in file boot/zephyr/include/target.h 

#if !defined(FLASH_DRIVER_NAME) || \
    !defined(FLASH_ALIGN) ||                  \
    !defined(FLASH_AREA_IMAGE_0_OFFSET) || \
    !defined(FLASH_AREA_IMAGE_0_SIZE) || \
    !defined(FLASH_AREA_IMAGE_1_OFFSET) || \
    !defined(FLASH_AREA_IMAGE_1_SIZE) || \
    !defined(FLASH_AREA_IMAGE_SCRATCH_OFFSET) || \
    !defined(FLASH_AREA_IMAGE_SCRATCH_SIZE)
#error "Target support is incomplete; cannot build mcuboot."
#endif

so anybody can give me some advise, thank you for your answer !

utzig, 2018-05-03:
You are required to have the 3 areas defined, but only `IMAGE_0` and `IMAGE_1` need to have the same size and sector config, they both would need to be 128k or 192k or whatever you need. `IMAGE_SCRATCH` can be a single sector, if your device supports 4k sectors you only need 4k for scratch.

Jeepgoing, 2018-05-10:
i have tried as your described，but when IMAGE_SCRATCH size is not equal to IMAGE_0, mcuboot will never to judge which slot to jump, it only can jump to slot0, also i have signed slot1 to upgrade.

Jeepgoing, 2018-05-10:
i have tried as your described，but when IMAGE_SCRATCH size is not equal to IMAGE_0, mcuboot will never to judge which slot to jump, it only can jump to slot0, also i have signed slot1 to upgrade.@utzig

Jeepgoing, 2018-05-10:
my target board have described flash partition in device tree, the problem is i do not want to use IMAGE_SCRATCH, i want to storage other things in this part, zephyr can support this, but mcuboot can not support. @nvlsianpu

nvlsianpu, 2018-05-10:
The issue might be that you flash driver doesn't provide API for fetch flash pages layout, or this API is disabled (recommended is Kconfig option CONFIG_FLASH_PAGE_LAYOUT=y). Because you didn't provide  FLASH_AREA_IMAGE_SECTOR_SIZE lvalue macro, you got following case https://github.com/runtimeco/mcuboot/blob/master/boot/zephyr/flash_map_legacy.c#L50 which cause the issue. As utzig mention you can have relatively smal scratch at size of one flash sector. But if you do not want this - it is possible to chose overwrite only mode for mcuboot - [CONFIG_BOOT_UPGRADE_ONLY=y](https://github.com/runtimeco/mcuboot/blob/master/boot/zephyr/Kconfig#L51)

Jeepgoing, 2018-05-11:
ok, thank you, i will try it.@nvlsianpu

Jeepgoing, 2018-05-14:
i solved this problem by modify https://github.com/runtimeco/mcuboot/blob/master/boot/zephyr/flash_map_legacy.c#L50
#define FLASH_AREA_IMAGE_SECTOR_SIZE FLASH_AREA_IMAGE_0_SIZE

closed 2018-05-14 by Jeepgoing