mcu-tools/mcuboot #278: AttributeError: '_RSAPrivateKey' object has no attribute 'sign'

zhaozhongchen, 2018-05-03:
I am want test usb dfu example in zephyr, but imgtool.py report that error. may be I installed the wrong version cryptography?

```
~/debug/port/github-os/mcuboot/scripts/imgtool.py sign \
> --key ~/debug/port/github-os/mcuboot/root-rsa-2048.pem \
> --header-size 0x200 \
> --align 8 \
> --version 1.2 \
> --included-header \
> ./zephyr/zephyr.bin \
> signed-zephyr.bin
Traceback (most recent call last):
  File "/home/qianfan/debug/port/github-os/mcuboot/scripts/imgtool.py", line 148, in <module>
    args()
  File "/home/qianfan/debug/port/github-os/mcuboot/scripts/imgtool.py", line 145, in args
    subcmds[args.subcmd](args)
  File "/home/qianfan/debug/port/github-os/mcuboot/scripts/imgtool.py", line 85, in do_sign
    img.sign(key)
  File "/home/qianfan/debug/port/github-os/mcuboot/scripts/imgtool/image.py", line 136, in sign
    sig = key.sign(bytes(self.payload))
  File "/home/qianfan/debug/port/github-os/mcuboot/scripts/imgtool/keys/rsa.py", line 92, in sign
    return self.key.sign(
AttributeError: '_RSAPrivateKey' object has no attribute 'sign'
```

utzig, 2018-05-03:
try reinstalling:

```
pip install --force-reinstall cryptography
```
I suspect this might be due to some native built module, which stops working after `OpenSSL` is updated locally.

zhaozhongchen, 2018-05-04:
seems i confused python2 and python3. I had changed my default py to python3 and reinstall cryptgraphy and its depends, it works