mcu-tools/mcuboot #280: zephyr: warn-on-open-flash-area feature (whenever flash_area_close become a real operation)

nvlsianpu, 2018-05-10:
When ever flash_are_close become non dummy function it is recommended to implement functionalities to notice whenever open/close calls were mismatched.

Such functionality is intended to be called just before do_boot().

mbolivar, 2018-05-16:
Thanks for logging this!

d3zd3z, 2019-12-02:
The Zephyr issue has merged, do we still need to do something on the MCUboot side?

mbolivar, 2019-12-03:
> The Zephyr issue has merged, do we still need to do something on the MCUboot side?

I'm not sure what this means. The open/close calls in zephyr still don't make it possible to see if there are areas which are still open right before do_boot() is called and issue a warning if so.

```c
int flash_area_open(u8_t id, const struct flash_area **fap)
{
	const struct flash_area *area;

	if (flash_map == NULL) {
		return -EACCES;
	}

	area = get_flash_area_from_id(id);
	if (area == NULL) {
		return -ENOENT;
	}

	*fap = area;
	return 0;
}

void flash_area_close(const struct flash_area *fa)
{
	/* nothing to do for now */
}
```

d3zd3z, 2019-12-03:
Where are you thinking of adding this? I would like to also add something comparable to the `flash_area_open` and `flash_area_close` in `sim/mcuboot-sys/csupport/run.c` that checks to make sure the calls are always matched. But, that might warrant a separate issue.

mbolivar, 2019-12-04:
> Where are you thinking of adding this? I would like to also add something comparable to the `flash_area_open` and `flash_area_close` in `sim/mcuboot-sys/csupport/run.c` that checks to make sure the calls are always matched. But, that might warrant a separate issue.

I'm not thinking of doing anything personally myself.

But the idea was that the zephyr implementations quoted above should be able to let mcuboot check if every open is matched by a close by the time do_boot() is called.

closed 2021-06-03 by github-actions[bot]