mcu-tools/mcuboot #286: ECDSA error when building on Zephyr

utzig, 2018-06-01:
I am unable to build using ECDSA on Zephyr.

```
mkdir build; cd build
cmake -DBOARD=nucleo_f401re ../
```

Running `make` here with RSA works ok. But after running `make menuconfig` and changing signature type to "Signature type (Elliptic curve digital signatures with curve P-256)" and running `make` again:

```
...
[ 57%] Building C object zephyr/ext/lib/crypto/mbedtls/CMakeFiles/ext__lib__crypto__mbedtls.dir/zephyr_init.c.obj
In file included from <command-line>:
/home/utzig/src/mynewt/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h:4:33: fatal error: mcuboot-mbedtls-cfg.h: No such file or directory
 #define CONFIG_MBEDTLS_CFG_FILE "mcuboot-mbedtls-cfg.h"
...
```

I tried fixing this by adding the `zephyr/include` to the build which than goes on to fail linking with conflict symbols against Zephyr's bundled `mbedts`.

utzig, 2018-12-17:
Your `prj.conf` needs both `CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256=y` and `CONFIG_BOOT_SIGNATURE_KEY_FILE="root-ec-p256.pem"` (whatever your EC key file is called), which will parse the PEM file and auto-generate the C code with those symbols you are missing. If your key file is RSA it will generate RSA symbols, and EC key files generates the symbols you are missing.

sigvartmh, 2019-03-04:
This issue is fixed right? It's a limitation in the prj.conf files where if one is set like the `CONFIG_BOOT_SIGNATURE_KEY_FILE` it can't be changed as the cmake function is already generated. So you will have to change `CONFIG_BOOT_SIGNATURE_KEY_FILE` in the `prj.conf`file and you can't do it `menuconfig` as it won't create the correct signing command since it uses the wrong file. With regards to the `mbedtls` couldn't there be a way to deselect `mbedtls` sources for zephyr when ecdsa is selected like `select MBEDTLS(=n)` when selecting `ecdsa`

closed 2019-03-04 by utzig