mcu-tools/mcuboot #307: Serial recovery on nRF requires a power-cycle after flashing

carlescufi, 2018-06-29:
When using the serial recovery mode and after flashing with `ninja flash` and resetting with button 1 pressed, `mcumgr` will time out unless one power-cycles the board.

Not urgent, should not block 1.2

carlescufi, 2018-06-29:
CC @lemrey

d3zd3z, 2018-06-29:
Any idea if this can be fixed in software, or does it just have to be documented?

carlescufi, 2018-06-29:
I think it should be fixable by software because the UART works fine for things like the console. We will look into this very soon, but it's not serious enough to warrant delaying the release.

lemrey, 2018-07-03:
It is possible this was an issue with the virtual machine (VirtualBox). The issue disappeared when switching from the EHCI 2.0 USB controller to the OHCI 1.1 USB controller; I was not able to reproduce this in VMWare player.

carlescufi, 2018-07-03:
@lemrey can you please run Logic16 with pca10056 to see if the request packet is actually reaching the nRF IC?

d3zd3z, 2019-12-02:
Is this still an issue?

carlescufi, 2019-12-02:
@nvlsianpu @lemrey is this still relevant?

d3zd3z, 2020-01-07:
Closing. Please re-open if this is still an issue.

closed 2020-01-07 by d3zd3z