mcu-tools/mcuboot #319: How the boot state exchanges?

BabaYB, 2018-07-24:
I have some questions about the state exchanging.    
First, is it right in the picture of the state exchanging when we download a new image and set the magic of SLOT1?    
So what will the next boot be?    
And if we set the image_ok in SLOT1 when the new app is running?Then the next boot will determines the state as BOOT_SWAP_TYPE_PERM, and swaping the old image in SLOT1 back to SLOT0?    
So the second is when shall we set and use the `boot_set_pending()`?   
[image]

BabaYB, 2018-07-24:
It would be better if there is a map of the state exchanging.

d3zd3z, 2019-12-02:
There have been quite a few updates to `docs/design.md`. Is your question answered there. If not, it would probably be best to answer this with a doc update.

d3zd3z, 2020-01-07:
Closing. Please open if there are aspects of this not described in the design doc.

closed 2020-01-07 by d3zd3z