mcu-tools/mcuboot #320: Bare metal sample code

kpyush, 2018-07-25:
Do you have any plans for providing the bare metal port in future..?

BabaYB, 2018-07-27:
You can take a look at the porting in [TF-M](https://git.trustedfirmware.org/trusted-firmware-m.git/), and also there are old commits with implement in zephyr on MK64F

kpyush, 2018-08-08:
Still I am unable to understand why OS binding is necessary in boot-loader, can someone help me understand?

utzig, 2018-08-08:
You could take a look here: https://github.com/runtimeco/mcuboot/blob/master/docs/PORTING.md

The main issue is the flash drivers required. If `mcuboot` should be ported to bare metal to a single board, which means one single flash driver, that would be OK to do but relying on the OS means having support for every board supported by the OS. No OS functionalities like scheduling are used, but some might be needed when doing crypto, although I think `tinycrypt` should be super easy to get running without an OS (if there's anything to do at all, also it's bundled with `mcuboot`). And answering your first question: not likely to happen because it's a lot work/maintainership and it doesn't seem like there are any obvious advantages (the target OSes are free and permissive so no reason not to use them)!

kpyush, 2018-08-10:
@utzig thanks for replying..understood your point.

closed 2018-08-14 by utzig

d3zd3z, 2018-08-14:
Probably the closest to "bare metal" would be to port MCUboot to CMSIS. This is more of a HAL than an RTOS. It provides flash drivers, and also the necessary code to initialize the hardware on startup. Some work has been done on this for trust firmware m, but has not been brought back into the main MCUboot tree.

utzig, 2018-08-14:
I think CMSIS only defines a Flash interface, one would still have to dig through the vendor's SDKs to get the implementation for all parts.

d3zd3z, 2018-08-14:
That depends on the device, and how much the vendor contributes to CMSIS. For example, for ARM devices, it isn't too difficult to make a bare-metal target with CMSIS. I wouldn't imagine a CMSIS port to be all that useful with a vendor that doesn't fully support CMSIS (which probably means with boards not made by Arm).