mcu-tools/mcuboot #324: mcuboot fails to compile due to missing intelhex python module

mnkp, 2018-08-07:
When compiling mcuboot for Zephyr following the instructions at [Building and using MCUboot with Zephyr](https://github.com/runtimeco/mcuboot/blob/master/docs/readme-zephyr.md)
I received the following error message
```
[13/194] Generating zephyr/autogen-pubkey.c
FAILED: zephyr/autogen-pubkey.c 
cd /home/piotr_m/work/mcuboot/boot/zephyr/build && /usr/bin/python3 /home/piotr_m/work/mcuboot/scripts/imgtool.py getpub -k /home/piotr_m/work/mcuboot/root-rsa-2048.pem > /home/piotr_m/work/mcuboot/boot/zephyr/build/zephyr/autogen-pubkey.c
Traceback (most recent call last):
  File "/home/piotr_m/work/mcuboot/scripts/imgtool.py", line 20, in <module>
    from imgtool import image
  File "/home/piotr_m/work/mcuboot/scripts/imgtool/image.py", line 21, in <module>
    from intelhex import IntelHex
ModuleNotFoundError: No module named 'intelhex'
```
`intelhex` python module is not included as the Zephyr build dependency in `scripts/requirements.txt`.

The mcuboot documentation should probably mentioned that this module needs to be installed by hand or it should be added to Zephyr dependencies in `scripts/requirements.txt`.

utzig, 2018-08-07:
It *is* included: https://github.com/runtimeco/mcuboot/blob/master/scripts/requirements.txt#L2

Are you running `master`?

mnkp, 2018-08-07:
Yes, `master`. But I didn't know I'm supposed to run 
```
cd ~/mcuboot  # your directory where mcuboot is cloned
pip3 install --user -r scripts/requirements.txt
```
from mcuboot. Documentation mentions only that mcuboot should be build as a standard Zephyr app, there is no mention of extra dependencies. These instructions should probably be added to the above mentioned "Building and using MCUboot with Zephyr" page?

utzig, 2018-08-07:
Ah, I see. It is documented on the `imgtool` doc but maybe adding a pointer from `readme-zephyr.md` could be a good idea. We are open to PRs! ;-)

mnkp, 2018-08-07:
Done: #325

closed 2018-08-08 by utzig