mcu-tools/mcuboot #326: allow slots with different alignment

nvlsianpu, 2018-08-08:
Objective is to handle the write/erase alignment being different between internal and external flash devices.

Looks like this shouldn't be not very complicated for combination of devices for which topical both alignment values are powers of 2. So it is possible to use last common multiples for both alignment values,

nvlsianpu, 2018-08-08:
This is about what is checked by boot_slots_compatible(void), and how swap operation is performed. flash_area_align API allow to cut form hardcode alignment size implementation for zephyr.

utzig, 2018-08-08:
`boot_slots_compatible` checks that both slots have the same number of sectors and that they have the same amount of sectors. What is easy relax here is to just allow slot1 having more sectors than slot0 (as long as they are all of the same size). Having differently sized sectors will be more complicated to due to the atomicity required by the swaps. Alignment should not be an issue because it is used mostly for metadata that stays in the slot and used by the same slot.

utzig, 2018-08-09:
For documentation purposes, copying some snippets of public slack discussion here:

> What is needed is to provide capability for operating with different sector size in slots. For instance slot_0 might be of 4 KB pages, while slot_1 might be of 16 KB pages.

> I think, boot_slots_compatible has to assure that scratch is able to fit the largest possible sector to be swapped and that every sector on the largest sector device has a size that is a multiple of the smaller one (as you mentioned)... should be possible to do in a way that is completely compatible with what already there... the devil might be in the details :wink: so nice to start by having a test case on the simulator with two slots that have those suggested sector sizes

@nvlsianpu I think I might need this as well, so I'll try to allow the simulator to already have different sector size support for slots!

utzig, 2019-01-09:
Initial support added with #359

closed 2019-01-09 by utzig