mcu-tools/mcuboot #333: Issue `Padding requested, but image does not start with zeros` even with `--pad`

Olivier-ProGlove, 2018-09-18:
I am trying to sign a Zephyr `hello_world` image but it does not work due to the error `Padding requested, but image does not start with zeros`

Steps to reproduce:

```
# Clone repositories
git clone git@github.com:zephyrproject-rtos/zephyr.git
git clone git@github.com:runtimeco/mcuboot.git

# Configure Zephyr
cd zephyr
source zephyr-env.sh

# Build image
mkdir build-hello && cd build-hello
cmake -DBOARD=nrf52840_pca10056 ../samples/hello_world
make

# Sign:
../../mcuboot/scripts/imgtool.py sign \
     --key ../../mcuboot/root-rsa-2048.pem \
     --header-size 0x200 \
     --align 8 \
     --version 1.0 \
     --slot-size 0x69000 --pad \
     ./zephyr/zephyr.hex signed.hex

Traceback (most recent call last):
  File "../../mcuboot/scripts/imgtool.py", line 181, in <module>
    imgtool()
  File "/home/olivier/.local/lib/python3.5/site-packages/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/home/olivier/.local/lib/python3.5/site-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/home/olivier/.local/lib/python3.5/site-packages/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/olivier/.local/lib/python3.5/site-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/olivier/.local/lib/python3.5/site-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "../../mcuboot/scripts/imgtool.py", line 139, in sign
    overwrite_only=overwrite_only)
  File "/tmp/temp/mcuboot/scripts/imgtool/image.py", line 87, in load
    obj.check()
  File "/tmp/temp/mcuboot/scripts/imgtool/image.py", line 121, in check
    raise Exception("Padding requested, but image does not start with zeros")
Exception: Padding requested, but image does not start with zeros
```

Same issue with/without `--pad` and `.bin`/`.hex`/`.elf`

utzig, 2018-09-18:
Not `--pad` related! Images built by the Zephyr build system are already padded with zeros on the place where the **header** will be written by `imgtool`. It's weird that yours don't seem to have it, but you can add it manually with `--pad-header`. Haven't you supplied the wrong `--header-size` maybe?

Olivier-ProGlove, 2018-09-18:
Here is the repositories versions I have been using (really the latest versions):

```
mcuboot$ git log -1
commit 42ad446fb0516d27dd387e477ef58d6d76589e2c
Author: Fabio Utzig <utzig@apache.org>
Date:   Tue Aug 14 08:55:23 2018 -0300

    Add `flash_area_erased_val` "mock" for Zephyr
    
    For Zephyr assume flash is always erased to 0xff. This allows backward
    compatibility of the new flash erased to 0 support.
    
    Signed-off-by: Fabio Utzig <utzig@apache.org>

zephyr$ git log -1
commit 8fc5fd55dec1f0808badc5508ca8cca497940f3e
Author: Andrei Emeltchenko <andrei.emeltchenko@intel.com>
Date:   Fri Sep 14 15:48:54 2018 +0300

    logger: Simplify UART backend initialization
    
    Move definition to sources of UART backend.
    
    Signed-off-by: Andrei Emeltchenko <andrei.emeltchenko@intel.com>
```

I have just done a clean build and dump the binary:

```
$ hexdump zephyr/zephyr.bin 
0000000 1b18 2000 1aa9 0000 1afd 0000 1935 0000
0000010 1935 0000 1935 0000 1935 0000 1935 0000
0000020 1935 0000 1935 0000 1935 0000 1571 0000
0000030 1935 0000 1935 0000 152d 0000 1935 0000
0000040 1a6d 0000 743b 0000 1a6d 0000 1a6d 0000
0000050 1a6d 0000 1a6d 0000 1a6d 0000 1a6d 0000
*
0000100 0000 0000 1271 0000 0000 0000 18a9 0000
0000110 0000 0000 18a9 0000 0000 0000 18a9 0000
*
0000130 0000 0000 96e5 0000 0000 0000 18a9 0000
0000140 0000 0000 18a9 0000 0000 0000 18a9 0000
0000150 0000 0000 18a9 0000 0000 0000 7401 0000
0000160 0000 0000 18a9 0000 2aec 2000 9a91 0000
0000170 0000 0000 18a9 0000 0000 0000 18a9 0000
0000180 0000 0000 18a9 0000 0000 0000 1399 0000
0000190 0000 0000 18a9 0000 0000 0000 18a9 0000
*
00001c0 0000 0000 7435 0000 0000 0000 18a9 0000
00001d0 0000 0000 18a9 0000 0000 0000 18a9 0000
*
0000280 b97b b972 2900 bfbe 2000 f04f 4100 e006
```

utzig, 2018-09-18:
Are you building with `samples/zephyr/Makefile` or defining `BOOT_HEADER_LEN`?

Olivier-ProGlove, 2018-09-18:
You are right I forgot `CONFIG_BOOTLOADER_MCUBOOT=y` (not present in Zephyr `hello_world` sample). Adding this value fixes my issue.

closed 2018-09-18 by Olivier-ProGlove