mcu-tools/mcuboot #351: Encrypted images should allow validation without access to secret/private key

utzig, 2018-10-19:
Currently an encrypted image is generated with validation TLVs (hash + sign) applied over the non-encrypted firmware data. It was pointed to me recently that a way of verifying without having to have access to the secret/private keys is desirable for situations where the image needs to be validated offline (like cloud, etc).

This requires adding extra information to assure validation is possible without having to decrypt or a way to use validation over encrypted data that maintains compatibility with the validation performed in the device.

d3zd3z, 2020-01-10:
This seems like we would need to have both validations, then. We don't really have an easy way, in either case, to verify that the other one would be correct, though. Most things that sign an encrypt sign the encrypted data, but that would be difficult to verify after the non-encrypted image is installed.

I'll keep thinking about it, but I don't yet have any better ideas.

tim-nordell-nimbelink, 2020-04-23:
It seems like there could be a SHA256 TLV added that is for the encrypted form of the payload. E.g. looking at https://juullabs-oss.github.io/mcuboot/design.html#protected-tlvs, you could have a TLV that hashes the A - B range of the encrypted version.  Additionally, a second TLV could be added (if wanted) that contains a signature for this new hash using a public key that the application has.

closed 2021-06-03 by github-actions[bot]