mcu-tools/mcuboot #371: Make imgtool available as a package.

jcarrano, 2018-12-14:
I'm  a maintainer for the [RIOT OS](https://github.com/RIOT-OS/RIOT) build system. We are using imgtool to sign mcuboot-compatible images.

Right now we have a _copy_ of [an old version of] imgtool. Upgrading means copying files and handling dependencies. It makes for big PRs and bloats the repo. It would be optimal if we could avoid that and instead download the tool. For that, imgtool would have to be packaged (with a setup.py and all that) and either:

- Put into PyPI, or..
- Put into a repo of its own, so that it can be installed with pip using a git url.

I can help you with the first part (the packaging) but of course I cannot push it to PyPI or put it in a separate repo (at least not without forking, and I don't want to do that).

utzig, 2018-12-17:
Good idea IMO, @carlescufi @d3zd3z @nvlsianpu what do you think?

utzig, 2019-01-09:
Initial support was added with #377

closed 2019-01-09 by utzig