mcu-tools/mcuboot #372: Run simulator tests with different header sizes

utzig, 2018-12-17:
PR #370 fixed an issue that happened when header sizes are larger than 32 bytes, which seems common place for some target OSes. The current simulator implementation always runs images with a header size equal to 32 bytes.

A simple option to configure the simulator image's header size at build time was added with commit https://github.com/JuulLabs-OSS/mcuboot/commit/e5831f6ed2914e347b7a7a109eab993d3add56fa. 

Maybe it would be useful to extend the simulator and travis script to enable tests to run for the most commonly used header sizes.

d3zd3z, 2019-01-18:
I think this mostly comes from Zephyr, where the vector table follows the header, and therefore must be aligned to a CPU-specific boundary. Common values will be powers of two, probably from 128 to 512 or so.

d3zd3z, 2019-12-02:
We should also test with larger values, as we found some problems from an unrelated bug that suggest there might be problems as this gets larger (say 1024). The challenge here is how to make the header size larger without having to do a cross-product test setup? We could make an environment variable or feature that does large sizes, but that wouldn't be tested regularly.

closed 2021-06-03 by github-actions[bot]