mcu-tools/mcuboot #373: Add an overview about supported features

jobrud, 2018-12-18:
Hi,
first I would like to thank you. Especially the possibility for reliable upgrades via image swap is very nice.
During my integration I switched between different algorithm for signing/encryption and I missed something like a overview which shows what is currently supported with which crypto library. 
I started to summarize the information which I found in the several docs:

## Image signing :
| crypto library  |RSA   |ECDSA |
| --------------- |:-----:|- ----:|
| tinycrypt       | no   | yes  |
| mbedTLS     | yes | no   |

## Image encryption key encryption:

(Note: Image will be encryted with AES-CTR-128)

| crypto library  |AES-KW-128 | RSA-OAEP |
| --------------- |:---------:| --------:|
| tinycrypt      | no        | no        |
| mbedTLS    | yes       | yes      |

## Imgtool support:

|             |      |
| ------------|:----:|
| RSA         | yes  |
| ECDSA       | yes  |
| AES-KW-128  | no   |
| RSA-OAEP    | yes   |

Is that correct ?

utzig, 2018-12-18:
Yes, that is correct!

d3zd3z, 2019-12-02:
Should this be added to the documentation?

utzig, 2019-12-03:
Could be added to the encrypted images document.

closed 2021-06-03 by github-actions[bot]