mcu-tools/mcuboot #382: imgtool sign creates invalid binary file from intel-hex file

nvlsianpu, 2019-01-04:
In case imgtool sign is called with intel-hex as input it produce binary file which looks like RAW intel-hex records was warped instead of binary.

step to reproduce:
```
./scripts/imgtool.py sign --key root-rsa-2048.pem --header-size 0x200 --align 8 --version 3.0 --slot-size 0x69000 zephyr.hex image.bin
```

utzig, 2019-01-07:
Yes, when input is HEX, output is HEX as well. Should only input be expected to be in HEX format? Or should there be a `--output-format` parameter?

nvlsianpu, 2019-01-07:
Can output format by recognized by file name extension. It is *.hex or *.bin. *.bin can be also the default one. I see this as more intuitive than parameter.

utzig, 2019-01-07:
Makes sense, @d3zd3z @carlescufi are you OK with changing this?

carlescufi, 2019-01-07:
Yes, I am OK as long as you can override it with a parameter because in some cases you might want to force it to one of the two particular formats.

utzig, 2019-01-07:
My understanding is that one should pass the last argument as `<filename>.hex` to output HEX formatted file, and output BIN otherwise! ;-)

carlescufi, 2019-01-07:
ah! true :) sorry I confused the input with the output.

SebastianBoe, 2019-01-07:
Assuming this is implemented, it would be great if this was documented in the CLI help. Currently it just says "outfile" and "infile", with no mention of which formats are supported (leading me to incorrectly believe yesterday that only .bin was supported).

utzig, 2019-01-07:
@SebastianBoe Updated the PR, now `-h` has the info:

```
Usage: imgtool.py create [OPTIONS] INFILE OUTFILE

  Create a signed or unsigned image

  INFILE and OUTFILE are parsed as Intel HEX if the params have .hex
  extension, othewise binary format is used

Options:
  -k, --key filename
  --align [1|2|4|8]          [required]
  -v, --version TEXT         [required]
  -H, --header-size INTEGER  [required]
...
```

nvlsianpu, 2019-01-09:
Was fixed in https://github.com/JuulLabs-OSS/mcuboot/pull/386

closed 2019-01-09 by nvlsianpu