mcu-tools/mcuboot #411: nRF51 doesn't build

SebastianBoe, 2019-01-30:
On nRF51, only 32kB is being set aside to MCUBoot, but MCUBoot needs about 40kB.

This results in a build error:

```
[201/206] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf 
: && ccache /home/sebo/Downloads/gcc-arm-none-eabi-8-2018-q4-major/bin/arm-none-eabi-gcc    zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -T zephyr/linker.cmd -Wl,-Map=/home/sebo/ncs/mcuboot/boot/zephyr/b/zephyr/zephyr.map -u_OffsetAbsSyms -u_ConfigAbsSyms -Wl,--whole-archive app/libapp.a zephyr/libzephyr.a zephyr/arch/arm/core/libarch__arm__core.a zephyr/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a zephyr/lib/libc/minimal/liblib__libc__minimal.a zephyr/ext/lib/crypto/mbedtls/libext__lib__crypto__mbedtls.a zephyr/drivers/flash/libdrivers__flash.a zephyr/drivers/serial/libdrivers__serial.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -L"/home/sebo/Downloads/gcc-arm-none-eabi-8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/thumb/v6-m/nofp" -L/home/sebo/ncs/mcuboot/boot/zephyr/b/zephyr -lgcc -Wl,--print-memory-usage -mthumb -mcpu=cortex-m0 -nostdlib -static -no-pie -Wl,-X -Wl,-N -Wl,--gc-sections -Wl,--build-id=none -Wl,--orphan-handling=warn -mabi=aapcs -march=armv6s-m && :
Memory region         Used Size  Region Size  %age Used
           FLASH:       39884 B        32 KB    121.72%
            SRAM:       24728 B        32 KB     75.46%
        IDT_LIST:          56 B         2 KB      2.73/home/sebo/Downloads/gcc-arm-none-eabi-8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: zephyr/zephyr_prebuilt.elf section `_TEXT_SECTION_NAME_2' will not fit in region `FLASH'
/home/sebo/Downloads/gcc-arm-none-eabi-8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 7116 bytes
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.

```

Environment:

zephyr: 416d39723335d0c8ff9f5f66771e09a3ca66e47f
mcuboot: 2bc2685
board: nrf51_pca10028

SebastianBoe, 2019-01-30:
NB: Removing the line

CONFIG_DEBUG=y

from prj.conf

brings us pretty close;

`           FLASH:       33292 B        32 KB    101.60%`

nvlsianpu, 2019-01-30:
Because of recently switching to new zephyr logger infrastructure, size with logger enabled increased. Logging can be disabled by CONFIG_BOOT_HAVE_LOGGING=n.

SebastianBoe, 2019-01-30:
That's great.

The bootloader should build out-of-the-box for all supported platforms, so we need to figure out some way to have CONFIG_BOOT_HAVE_LOGGING=n when nrf51.

Long term, it might be possible to do;

```
config BOOT_HAVE_LOGGING
	bool "MCUboot have logging enabled"
	default n if DT_FLASH_SIZE < (32 * 1024)
	default y
	select LOG
	select LOG_INPLACE_PROCESS
	help
	  If y, enables logging on the serial port. The log level can
	  be defined by setting `LOG_DEFAULT_LEVEL`.
	  If unsure, leave at the default value.
```

PR #412 (2019-01-30): zephyr:board: squash nrf51_pca10028 flash footprint
Disable logging and enable size optimizations on
nrf51_pca10028 target in order to fit in boot slot size
for out-off-the-box build.

fixes #411
commit e82e3163 (2019-01-30): zephyr:board: squash nrf51_pca10028 flash footprint

Disable logging and enable size optimizations on
nrf51_pca10028 target in order to fit in boot slot size
for out-off-the-box build.

fixes #411
closed 2019-01-30 by nvlsianpu