mcu-tools/mcuboot #417: Question about MCUboot

Laczen, 2019-02-05:
Hi, I have been looking at MCUboot and I like the project.

I have a very basic question: Would it not be advantageous to add the place where an image is expected to be placed in the image header ?

It would remove the need for flash_map,
It would allow for libraries to be compiled and placed where they are expected to be,

MCUboot would only need to know two locations:
a. Where do I jump to at the end of MCUboot
b. Where are the updates located

It would also allow easy extension for non-XIP processors: at the end of MCUboot it jumps to a secondary bootloader that copies the image(s) that are placed at the correct location by mcuboot to ram.

nvlsianpu, 2019-02-05:
My concerns on your proposition are:
- upload image with bad address might  brick the device. This might happened even accidentally for properly signed image (by human fault). So mcuboot needs to keep boundary info anyway.
- multi-mage suport is in progress, so using flash_map abstraction becomes fruitfull.
- flash_area primitives are porting interface, thanks them switching to use of internal and external flash device is very easy.
-flash_map might become dynamic entity at some point, why to cutt this option out

Laczen, 2019-02-05:
@nvlsianpu, Thanks for your reply.

I agree that checks on the upload image including its address are required. Checks that directly pop up:
1. Don't allow the bootloader to be overwritten (or maybe not, if its executed from ram).
2. Don't allow updates that partially overwrites libraries (unless you are overwriting the lib itself)

My major concern with flash_map is that it is something that should be written in flash at bootloader programming. At the moment it is generated by both the app and the bootloader at compile time, if they are not in sync you can also brick the device (by human fault). Only when it resides in flash and can be read by the bootloader and the app it makes sense.

Making flash map dynamic would really scare me: updating the flash map would result in the impossibility to restore a previous version.

closed 2019-03-24 by Laczen