mcu-tools/mcuboot #418: MCUboot on devices with limited flash

Laczen, 2019-02-07:
On small flash devices (e.g. bbc microbit or nrf82810) I cannot get MCUboot working because the swap method limits the amount of flash available for a image to small (my app for a bt mesh device is at least 119kB).

Is it possible to extend mcuboot with a bt smp interface and only to allow "in-place" upgrades (like serial does now) ?

nvlsianpu, 2019-02-07:
looks like it is possible. I see here two independent feature - one is only-image update using serial recover. This shouldn't be so hard to achieve. Worth to mention that it will required signature verification prior the update.

Second is addition of BT for serial recovery - this is more complicated. Need to port service from zephyr or my-newt (hopefully with code simplification as dfu needs only smal subset of commands). 

Generally I think that addition of this feature make sense.

utzig, 2019-02-07:
You can try setting your slot1 size to 0 (and slot0 to all remaining free space!); this should fail on slot compatibility checks and just boot slot0. You can also enable overwrite-only mode to make the bootloader smaller. This should allow in-place serial upgrade. For BLE there is no support but follow @nvlsianpu suggestion above.

Laczen, 2019-02-07:
Cool, 
Regarding the signature verification, how is this done for the serial upgrade (now, without any changes)? Does mcuboot do the verification after loading to slot0 (before starting slot 0) or does it do this before loading to slot0 ?

utzig, 2019-02-07:
> Regarding the signature verification, how is this done for the serial upgrade (now, without any changes)? Does mcuboot do the verification after loading to slot0 (before starting slot 0) or does it do this before loading to slot0 ?

No validation for serial upgrade; packets are just written to sectors, validation would need the whole image to be uploaded first. Validation on following boots is done if you have `MCUBOOT_VALIDATE_SLOT0` defined, otherwise no validation is done.

Laczen, 2019-02-07:
Hi, my idea is to start from the smp_svr example in zephyr as this already has the required routines to load to flash. I would add the image verification and boot into slot 0 if the image is verified. I still need to figure out how to keep it in the bootloader when needed, a possible solution would be to mess up the image trailer so verification fails.

closed 2019-03-24 by Laczen