mcu-tools/mcuboot #421: Question about the new changes

chtavares592, 2019-02-12:
Hello, I have a question, if anyone can help me I will be very happy. I have an application on zephyr, that downloads an image and saves it in flash area 1, and before the last modifications I just rebooted the board and it already comes out running the downloaded image. Now, when I rebooted happens that error bellow:

```
[00:00:00.004,948] <inf> mcuboot: Starting bootloader
[00:00:00.010,967] <inf> mcuboot: Image 0: magic=good, copy_done=0x3, image_ok=0x1
[00:00:00.018,793] <inf> mcuboot: Scratch: magic=unset, copy_done=0x5b, image_ok=0x3
[00:00:00.016,952] <inf> mcuboot: Boot source: slot 0
[00:00:00.014,445] <inf> mcuboot: Swap type: test
***** BUS FAULT *****
  Imprecise data bus error
***** Hardware exception *****
Current thread ID = 0x00000000
Faulting instruction address = 0x3ebe
```

What change should I make so that the board keeps restarting in flash area 1 ??? Thank you

utzig, 2019-02-12:
Are you running zephyr and mcuboot `master`? (you should probably sync both together). In case you are, is it possible to find out what change started causing this? (`git bisect`might be useful). And check what's located at `0x3ebe`...

chtavares592, 2019-02-12:
Hello, I found where started the error on mcuboot. The error started here on Zephyr: 

```
commit dc0713a706e7e2898c172c6c0965b1b784859841
Author: Andy Ross <andrew.j.ross@intel.com>
Date:   Thu Jul 5 12:50:41 2018 -0700

    kernel: Cleanup. Remove redundant test when calling _Swap()
    
    _Swap() must already handle the case where _get_next_ready_thread() is
    the same as _current.
    
    Signed-off-by: Andy Ross <andrew.j.ross@intel.com>

```

 Unfortunately, I don't know how to solve it.

utzig, 2019-02-12:
This looks like a recently merged change in the scheduler; mcuboot runs with multithreading disabled, so could be a side-effect. @nvlsianpu any ideas?

nvlsianpu, 2019-02-12:
No Idea - unfortunately I don't have time to investigate this now.
cc @carlescufi

d3zd3z, 2019-02-12:
I'll investigate. The zip of Zephyr isn't booting into MCUboot at all for me, although I'm not sure if that is a logging change or this problem.

SebastianBoe, 2019-02-13:
> The tip of Zephyr isn't booting into MCUboot at all for me

For me neither, I believe this is https://github.com/zephyrproject-rtos/zephyr/issues/13249

closed 2019-03-08 by chtavares592