mcu-tools/mcuboot #427: Image swap status update issue

davidvincze, 2019-02-20:
When the number of status fields is equal to the value of BOOT_STATUS_MAX_ENTRIES the `swap_size` field (in the trailer) is overwritten during image swapping. Currently the max. number of status fields :
```c
#define BOOT_STATUS_MAX_ENTRIES    BOOT_MAX_IMG_SECTORS
```
 In the pictures below we can see a case when the swapping operation consists of one step only, the BOOT_STATUS_MAX_ENTRIES=1 (the swap status area contains only 1 status field) therefore all the status fields will be used. In another case if only one flash sector can fit into the scratch area and the number of sectors of any of the images is equal to BOOT_MAX_IMG_SECTORS the situation is the same, all the status fields will be used.
[image]
[image]

Image trailer status area start address: 0x1017FFD5
step.1: the bootloader records the state0 -> state1 transition (writes 0x2)
step.2: the bootloader records the state1 -> state2 transition (writes 0x3)
step.3: the bootloader records the state2 -> state0 transition (writes 0x1,
    transition to another status field, the second image)

In the second picture we can see as the bootloader overwrites the `swap_size` field (0x000A0240) with 0x1 (write size = 1byte) in step.3. Therefore I would suggest a modification to __record the completion of a state__ and not the transition between two states (the current behaviour). This way the bootloader won't write into the next status field in step.3 when there will be no more `boot_swap_sectors()` operations (and like in this case there are no more status fields).

```c
            | rec0 | rec1 | rec2 | Image locations
    --------+------+------+------+------------------------------
    state 0 | 0xff | 0xff | 0xff | S0:Image0 S1:Image1 Scr:n/a
    state 1 | 0x01 | 0xff | 0xff | S0:Image0 S1:n/a    Scr:Image1
    state 2 | 0x01 | 0x02 | 0xff | S0:n/a    S1:Image0 Scr:Image1
    state 3 | 0x01 | 0x02 | 0x03 | S0:Image1 S1:Image0 Scr:n/a
```

solsbarry, 2019-09-04:
Based on my testing. David is write, and if you do the changes I proposed above, the result matches the documentation, and fixes David's issue. However it breaks the resuming an incomplete swap functionality. To fix it, you also need to make the following change or some equivalent.
In
`boot_read_status_bytes()`
change
```
if (found) {
        if (!found_idx) {
            found_idx = i;
        }
        found_idx--;  // index of last record written.
        bs->idx = (found_idx / BOOT_STATUS_STATE_COUNT) + 1;      // index (starting at 1) where state was found
        bs->state = (found_idx % BOOT_STATUS_STATE_COUNT) + 1;    // state found       
    }
```
to
```
if (found) {
        if (!found_idx) {
            found_idx = i;
        }
        found_idx--;  // index of last record written.
        bs->idx = (found_idx / BOOT_STATUS_STATE_COUNT) + 1;      // index (starting at 1) where state was found
        bs->state = (found_idx % BOOT_STATUS_STATE_COUNT) + 1;    // state found

        // If state 2 has been completed then bump the index up one. and change the state to 0.
        if (bs->state == BOOT_STATUS_STATE_2)
            {
            bs->idx++;
            bs->state = BOOT_STATUS_STATE_0;
            }
        else{           // otherwise just bump the state up one since that is the state that needs to be completed next.
            bs->state++;
            }
    }
```

The comments I have added to the code above assume the changes I described in my previous post have been made.

I successfully tested this change on a situation where a partially completed swap operation was resumed on boot-up. I think the old code worked fine in every situation except the one David describes. But these changes fix that and help the code match the documentation.

d3zd3z, 2019-09-04:
Some quick sim tests, by increasing the size of the image, I see several failures, depending on the size.

1. In one instance, boot_go just returns 3 (`BOOT_EBADIMAGE`).
2. Other times, I've seen an assertion failure right after the call to `boot_read_swap_size` in `loader.c`.
3. I've also gotten a failure trying to write to unerased data. I'm assuming that this means that my image was large enough to overwrite part of the the trailer.

It is a little challenging to figure out how big to make the image, since the TLVs can be different sizes, depending on what is in them. Also, making the images large tends to make the whole simulator run very slowly. It'd be nice to test these, so maybe we can have some special configuration that tests them in some cases, or just an option to test them manually.

utzig, 2019-09-05:
Commit https://github.com/JuulLabs-OSS/mcuboot/pull/533/commits/290393f6e89f5f733a550f9622ee67d5996ef91b should fix the issue related initially by @davidvincze. I am working on updating the documentation according to the way things work today.

davidvincze, 2019-09-11:
Yes it does, thank you for the patch! I tested this new swap status control mechanism and it works as I expected. :) I'm closing this issue...

closed 2019-09-11 by davidvincze