mcu-tools/mcuboot #428: MBUboot build fails if serial recovery enabled

ChristopherBittner, 2019-02-22:
Im trying to build it for nrf52840_pca10059, serial recovery is enabled in that configuration by default.

First try error:
```
../serial_adapter.c:196:32: error: 'CONFIG_CDC_ACM_PORT_NAME' undeclared (first use in this function); did you mean 'CONFIG_CDC_ACM_PORT_NAME_0'?
  uart_dev = device_get_binding(CONFIG_CDC_ACM_PORT_NAME);
                                ^~~~~~~~~~~~~~~~~~~~~~~~
                                CONFIG_CDC_ACM_PORT_NAME_0
```

After applying suggested value i.e. _0 suffix:

FAILED: zephyr/zephyr_prebuilt.elf 
```
: && ccache /home/chbi/gcc/bin/arm-none-eabi-gcc    zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -T zephyr/linker.cmd -Wl,-Map=/home/chbi/ncs/mcuboot/boot/zephyr/build/zephyr/zephyr.map -u_OffsetAbsSyms -u_ConfigAbsSyms -Wl,--whole-archive app/libapp.a zephyr/libzephyr.a zephyr/arch/arch/arm/core/libarch__arm__core.a zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a zephyr/arch/arch/arm/core/cortex_m/mpu/libarch__arm__core__cortex_m__mpu.a zephyr/lib/libc/minimal/liblib__libc__minimal.a zephyr/boards/arm/nrf52840_pca10059/libboards__arm__nrf52840_pca10059.a zephyr/ext/lib/crypto/mbedtls/libext__lib__crypto__mbedtls.a zephyr/ext/lib/encoding/tinycbor/libext__lib__encoding__tinycbor.a zephyr/drivers/gpio/libdrivers__gpio.a zephyr/drivers/flash/libdrivers__flash.a zephyr/drivers/serial/libdrivers__serial.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -L"/home/chbi/gcc/bin/../lib/gcc/arm-none-eabi/7.3.1/thumb/v7e-m" -L/home/chbi/ncs/mcuboot/boot/zephyr/build/zephyr -lgcc -Wl,--print-memory-usage -mthumb -mcpu=cortex-m4 -nostdlib -static -no-pie -Wl,-X -Wl,-N -Wl,--gc-sections -Wl,--build-id=none -Wl,--orphan-handling=warn -mabi=aapcs -march=armv7e-m && :
Memory region         Used Size  Region Size  %age Used
           FLASH:       60488 B         1 MB      5.77%
            SRAM:       33376 B       256 KB     12.73%
        IDT_LIST:          88 B         2 KB      4.30zephyr/libzephyr.a(boot_serial.c.obj): In function `boot_serial_in_dec':
/home/chbi/ncs/mcuboot/boot/boot_serial/src/boot_serial.c:591: undefined reference to `min'
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.

```

utzig, 2019-02-22:
#425 should fix this.

ChristopherBittner, 2019-02-22:
Yes, looks like it is addressing everything mentioned above. Thanks.

closed 2019-02-22 by ChristopherBittner