mcu-tools/mcuboot #429: Which version of zephyr is compatible with v1.3 mcuboot release

nvlsianpu, 2019-02-22:
### Problem:
Can not to build mcuboot v1.3 against zephyr 1.13 and  zephyr master HEAD.

### Solution a)
If you want to use v1.3 without any patch use the last compatible zephyr version.

build mcuboot against last full compatible zephyr mainline SHA:  https://github.com/zephyrproject-rtos/zephyr/commit/11ca2b61339373b2e0e5b4e0505b1ea1f53747cf

It is also possible to build until newer zephyr SHA: https://github.com/zephyrproject-rtos/zephyr/commit/85c8c0fe3cffeec929531fa2a42d8dfdfbcf71d7 but loggs stops working after [zephyr 11cab6](https://github.com/zephyrproject-rtos/zephyr/commit/11ca2b61339373b2e0e5b4e0505b1ea1f53747c)

### Solution b)
If you can accept a few aligning patch on v1.3 then switch to mcuboot mainline version use [mcuboot  ver. redy for zephyr 1.14 zephyr](https://github.com/JuulLabs-OSS/mcuboot/commit/0cd178d0e9e4cfe5c5ff6d62fba1b7c072138da7) or newer.

zephyr master HEAD of this note time i https://github.com/zephyrproject-rtos/zephyr/commit/c2d5e7b01ccff8be61ca139f578fdbce84b08713

PhilippFinke, 2019-04-30:
I don't think that this is a solution to my problem. I can build mcuboot v1.3.0 against Zephyr v1.14.0. The problem is that the flash map is incorrectly arranged. The problem even exists when I build a standalone version of my application. The array default_flash_map (that is used by mcuboot to retrieve the slot information) looks like this in the debugger.

default_flash_map[0] = <info of mcuboot partition>
default_flash_map[1] = <info of slot 1 partition>
default_flash_map[2] = <info of slot 0 partition>
default_flash_map[3] = <info of scratch partition>
default_flash_map[4] = <info of storage partition>

mcuboot expects the info of slot 0 on array index 1, but there is the info of slot 1.

nvlsianpu, 2019-04-30:
mcuboot 1.3.0 is not compatible with zephyr 1.14.0. Use mcuboot master instead. What you observed is dynamic generation of the flash_map items, therefore you can expect any partitions on any index in the array.

PhilippFinke, 2019-04-30:
Thanks for your reply. This fixed my problem. I thought the flash_map is ordered, but good to know that the partitions can be generated in any order.

d3zd3z, 2019-12-02:
These particular version incompatibility issues should be past. It would be good to define a Zephyr release as part of a release of MCUboot (and say, make sure it works with the last Zephyr LTS). It is also reasonable that tip of MCUboot work with tip of Zephyr, not withstanding bugs that need to be fixed.

closed 2019-12-02 by d3zd3z