mcu-tools/mcuboot #433: Constrained devices don't build out-of-the-box with Zephyr

SebastianBoe, 2019-02-28:
Constrained devices don't build out-of-the-box with Zephyr.

For instance, bbc_microbit doesn't build.

We have solved this for specific boards, e.g. #411 .

But now that DT symbols can be accessed from Kconfig, it should be possible to turn off unnecessary features based on FLASH / RAM size and solve this generally.

d3zd3z, 2019-12-02:
Is disabling logging typically enough for this? Not having logging on does make MCUboot more difficult to debug, though. Can we solve this by means other than documentation? e.g. Is there a mechanism to have board-specific Kconfigs in the app directory (like we do with device tree overlays)?

SebastianBoe, 2019-12-03:
>  e.g. Is there a mechanism to have board-specific Kconfigs in the app directory

Yes, see for instance https://github.com/JuulLabs-OSS/mcuboot/pull/412/files

but it would scale better to use generic DT information like flash and ram size to set Kconfig options. This is possible by defining a new Kconfig root with additional intializations of the options that depend on DT size information.

closed 2021-06-03 by github-actions[bot]