mcu-tools/mcuboot #444: MCUBoot is not placed in proper partition - Zephyr nrf52840_pca10059

MarekPieta, 2019-03-15:
MCUBoot is not placed in mcuboot flash partition on Zephyr nrf52840_pca10059 (tested with default configuration).
It is placed in the beginning of flash even if DTS specifies otherwise (according to current DTS it should be placed after Nordic MBR).

utzig, 2019-03-15:
Are you using a release or running from `master`? If so what is your latest sha?

MarekPieta, 2019-03-15:
I'm using MCUBoot from repo: NordicPlayground/fw-nrfconnect-mcuboot
commit sha: 59fde9c792bfaa36887c860fb8cd0ca1f1bc4db5

nvlsianpu, 2019-03-15:
cc @lemrey

lemrey, 2019-03-15:
Right, something seems wrong; will open a PR to fix this later today.

nvlsianpu, 2019-03-15:
Looks like this can be a case for any else target.

nvlsianpu, 2019-03-19:
fixed via https://github.com/zephyrproject-rtos/zephyr/pull/14620 + #447

closed 2019-03-19 by nvlsianpu