mcu-tools/mcuboot #451: Verifying Image Flash Location

jwknaup, 2019-03-26:
I am trying to follow the tutorial at https://mcuboot.com/mcuboot/readme-zephyr.html and the information on zephyr's website to flash an image to a nrf52 board. I successfully flashed MCUboot boot and can see it working in the console. However, when I flash the hello-world sample image, it appears to overwrite MCUboot and There is no output in the console.

nvlsianpu, 2019-03-27:
https://docs.zephyrproject.org/latest/guides/device_mgmt/dfu.html
> Your application’s .conf file needs to enable the CONFIG_BOOTLOADER_MCUBOOT Kconfig option in order for Zephyr to be built in an MCUboot-compatible manner

so add config mentioned above (`CONFIG_BOOTLOADER_MCUBOOT=y`) to your project conf file, and generate the project.

jwknaup, 2019-03-27:
It is set and still doesn't work

nvlsianpu, 2019-03-28:
Did you sign the image? https://docs.zephyrproject.org/latest/samples/subsys/mgmt/mcumgr/smp_svr/README.html#step-4-sign-the-image

jwknaup, 2019-03-28:
Yes, looking at the linker script, the flash offset is not set in spite of the CONFIG_BOOTLOADER_MCUBOOT option.

nvlsianpu, 2019-03-29:
I used flowing command to build the project (from the parent path to zephyr directory)
` west build -d ./build_hw --board nrf52_pca10040  ./zephyr/samples/hello_world/ -- -DCONFIG_BOOTLOADER_MCUBOOT=y`

This works for me  well: Hello Word get linked into code partition (so beginning form 0xC000, with offset of 0x200.)

Are you sure that you are using recent zephyr master?

d3zd3z, 2019-12-02:
Any updates on this? @jwknaup are you able to get this working?

d3zd3z, 2020-01-07:
Please reopen if this is still an issue.

closed 2020-01-07 by d3zd3z