mcu-tools/mcuboot #474: How to use MCUboot for delta Firmware update OTA ?

AnushaKamat, 2019-04-29:
Where does the the difference file between newer and older version of the firmware go sit in memory ? And does MCUboot handle the patching of older firmware and diff ?

sigvartmh, 2019-05-09:
MCUBoot does not support delta patching

mbolivar, 2019-05-12:
Hi @AnushaKamat, what @sigvartmh is true, but that does not mean you can't implement a delta-based approach in an MCUboot-based system.

However, since MCUboot doesn't support this (which is by design to avoid introducing extra complexity into a what is meant to be an immutable bootloader), you need to find a way to apply firmware diffs to your current firmware image (in slot 0) to an update image (in slot 1). Then reboot into MCUboot as usual. This allows you to send firmware diffs over the network as long as your application can apply them to itself without adding extra complexity to MCUboot itself.

@d3zd3z @utzig I think this should be an item on a new FAQ system in the docs or on mcuboot.com. Another common one I see is about firmware images in external flash, like SPI flash.

AnushaKamat, 2019-05-13:
Thank you all for your response. I wanted to be sure if there isn't an option provided by MCUboot before handling it in the application. Makes it clear.

closed 2019-05-13 by AnushaKamat

niondir, 2022-09-16:
It's a quiet old question but I'm working on exactly this.

My main issue is the image trailer. MCUBoot is modifying the trailer (e.g. Confirmation flag) in the flash. Thus the firmware in flash differs from the original firmware applied. A potential patch need some knowledge about the trailer to be applied correctly. 
Maybe someone reads this and had similar issues in the past. I'm interested in solutions for the problem.

zeineebou, 2024-07-17:
Hello @niondir, it's an old subject but I'm working on this now and I would like to know how did resolve this issue?

niondir, 2024-07-19:
@zeineebou I actually did not finally solved it.

But the plan was:
I copy the running Firmware over into a region where the patch will be applied. During that copy I would restore the original trailer so the patch can work as intended.

But that needs good knowledge about the trailer and it's structure.

atif-maqsood, 2024-09-16:
@zeineebou  I hope you have managed to resolve this? Have you used any specific libraries to enable the delta firmware update or it is full implementation in the application developed by yourself?