mcu-tools/mcuboot #480: Watchdog reset during bootloader stage breaks image swap

MariuszSkamra, 2019-05-16:
mcuboot version: 1.2.0 + 0001-MCUBOOT-Config-file-changes.patch
Zephyr version: 1.13.0 + 0001-ZEPHYR-Add-watchdog-to-smp_svr-sample.patch
Board: nrf52840-Preview-DK
Zephyr SDK: 0.9.5

I enclose binaries, so that one can build yourself or use mine.
In the patch I've added Watchdog to Zephy's smp_srv sample.

Steps to reproduce:

1. Build boot, application and sign it.
```
$ ./scripts/imgtool.py sign --key root-ec-p256.pem --slot-size 0x69000 --header-size 0x200 --align 8 --version 1.0.0 ../zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/zephyr.hex zephyr_signed.hex
```
```
$ mergehex -m boot/zephyr/build/zephyr/zephyr.hex zephyr_signed.hex -o zephyr_boot_signed.hex
```

2. Chiperase and flash:

```
$ nrfjprog --program zephyr_boot_signed.hex --chiperase --reset
Parsing hex file.
Erasing user available code and UICR flash areas.
Applying system reset.
Checking that the area to write is not protected.
Programing device.
Applying system reset.
Run.
```

3. Verify
```
$ mcumgr --conntype ble --connstring 'peer_name=Zephyr' image list
Images:
 slot=0
    version: 1.0.0
    bootable: true
    flags: active confirmed
    hash: 36699217adc9ff5c08a66cd0845a187830977528ccab345480fac6fa54d9c67f
Split status: N/A (0)
```
```
$ nrfjprog --memrd 0x74e58 --n 424
0x00074E58: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074E68: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074E78: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074E88: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074E98: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074EA8: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074EB8: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074EC8: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074ED8: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074EE8: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074EF8: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074F08: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074F18: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074F28: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074F38: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074F48: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074F58: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074F68: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074F78: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074F88: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074F98: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074FA8: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074FB8: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074FC8: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074FD8: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074FE8: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF   |................|
0x00074FF8: FFFFFFFF FFFFFFFF                     |........|`
```

4. Bump version
```
$ ./scripts/imgtool.py sign --key root-ec-p256.pem --slot-size 0x69000 --header-size 0x200 --align 8 --version 1.0.1 ../zephyrproject/zephyr/samples/subsys/mgmt/mcumgr/smp_svr/build/zephyr/zephyr.bin zephyr_signed.bin
```

5. Update the firmware via mcumgr
```
[root@localhost smart-tag]# mcumgr --conntype ble --connstring 'peer_name=Zephyr' image upload /home/mskamra/projects/mcuboot/zephyr_signed.bin
 111.05 KiB / 111.05 KiB [========================================================================================================================================================================================================================================] 100.00% 3.45 KiB/s 32s
Done
[root@localhost smart-tag]# mcumgr --conntype ble --connstring 'peer_name=Zephyr' image list
Images:
 slot=0
    version: 1.0.0
    bootable: true
    flags: active confirmed
    hash: 36699217adc9ff5c08a66cd0845a187830977528ccab345480fac6fa54d9c67f
 slot=1
    version: 1.0.1
    bootable: true
    flags: 
    hash: 99fcb7f25aae5a7657fbf8b2f9a07c464efbc3d6d177ade2ece9b18c3d5ce7ba
Split status: N/A (0)
[root@localhost smart-tag]# mcumgr --conntype ble --connstring 'peer_name=Zephyr' image test 99fcb7f25aae5a7657fbf8b2f9a07c464efbc3d6d177ade2ece9b18c3d5ce7ba
Images:
 slot=0
    version: 1.0.0
    bootable: true
    flags: active confirmed
    hash: 36699217adc9ff5c08a66cd0845a187830977528ccab345480fac6fa54d9c67f
 slot=1
    version: 1.0.1
    bootable: true
    flags: pending
    hash: 99fcb7f25aae5a7657fbf8b2f9a07c464efbc3d6d177ade2ece9b18c3d5ce7ba
Split status: N/A (0)
```

6. Observe the confirmed flag that stays in slot 1 every reset.
```
$ mcumgr --conntype ble --connstring 'peer_name=Zephyr' reset
Done
[root@localhost smart-tag]# mcumgr --conntype ble --connstring 'peer_name=Zephyr' image list
Images:
 slot=0
    version: 1.0.1
    bootable: true
    flags: active
    hash: 99fcb7f25aae5a7657fbf8b2f9a07c464efbc3d6d177ade2ece9b18c3d5ce7ba
 slot=1
    version: 1.0.0
    bootable: true
    flags: confirmed
    hash: 36699217adc9ff5c08a66cd0845a187830977528ccab345480fac6fa54d9c67f
Split status: N/A (0)
[root@localhost smart-tag]# mcumgr --conntype ble --connstring 'peer_name=Zephyr' reset
Done
[root@localhost smart-tag]# mcumgr --conntype ble --connstring 'peer_name=Zephyr' image list
Images:
 slot=0
    version: 1.0.0
    bootable: true
    flags: active
    hash: 36699217adc9ff5c08a66cd0845a187830977528ccab345480fac6fa54d9c67f
 slot=1
    version: 1.0.1
    bootable: true
    flags: confirmed
    hash: 99fcb7f25aae5a7657fbf8b2f9a07c464efbc3d6d177ade2ece9b18c3d5ce7ba
Split status: N/A (0)
[root@localhost smart-tag]# mcumgr --conntype ble --connstring 'peer_name=Zephyr' reset
Done
[root@localhost smart-tag]# mcumgr --conntype ble --connstring 'peer_name=Zephyr' image list
Images:
 slot=0
    version: 1.0.1
    bootable: true
    flags: active
    hash: 99fcb7f25aae5a7657fbf8b2f9a07c464efbc3d6d177ade2ece9b18c3d5ce7ba
 slot=1
    version: 1.0.0
    bootable: true
    flags: confirmed
    hash: 36699217adc9ff5c08a66cd0845a187830977528ccab345480fac6fa54d9c67f
Split status: N/A (0)
```

I enclose the binaries, patches and memory dump 
[wdt_issue.zip](https://github.com/JuulLabs-OSS/mcuboot/files/3187230/wdt_issue.zip)

Here you can observe the peaks after 5 seconds that indicate WDT reset SoC.
[image]

MariuszSkamra, 2019-05-16:
@utzig FYI

MariuszSkamra, 2019-05-16:
I've checked this with newest Zephyr master (zephyr-v1.14.0-834-ga2fa8e9c1c) and newest (MCUboot v1.3.0-68-g272702a) and this issue still exists.

```
$ mcumgr -i 1 --conntype ble --connstring 'peer_name=Zephyr' image list
Images:
 slot=0
    version: 1.0.0
    bootable: true
    flags: active
    hash: a84f1b4ef6222e278b8b15455796ed4ae8ac4e87f45357baf8b40832c2317cf2
 slot=1
    version: 1.0.1
    bootable: true
    flags: confirmed
    hash: 1e729cfdba73fb8af0cfd2a781512c9d3b61d3dd6d3214d5025a09d7be4b4a56
Split status: N/A (0)
$ mcumgr -i 1 --conntype ble --connstring 'peer_name=Zephyr' reset
Done
$ mcumgr -i 1 --conntype ble --connstring 'peer_name=Zephyr' image list
Images:
 slot=0
    version: 1.0.1
    bootable: true
    flags: active
    hash: 1e729cfdba73fb8af0cfd2a781512c9d3b61d3dd6d3214d5025a09d7be4b4a56
 slot=1
    version: 1.0.0
    bootable: true
    flags: confirmed
    hash: a84f1b4ef6222e278b8b15455796ed4ae8ac4e87f45357baf8b40832c2317cf2
Split status: N/A (0)
```
See the patches and binaries attached

[wdt_issue_new_sources.zip](https://github.com/JuulLabs-OSS/mcuboot/files/3187433/wdt_issue_new_sources.zip)

utzig, 2019-05-16:
@MariuszSkamra I just checked that Zephyr uses 4 as write-size of nrf52840, so you need to correct the flash dump address and size, the correct command would be: `nrfjprog --memrd 0x749d8 --n 1576`

MariuszSkamra, 2019-05-17:
Here's what I've got when I enabled debug logs and image validation (`CONFIG_BOOT_VALIDATE_SLOT0=y`, `CONFIG_LOG_DEFAULT_LEVEL=4`)

-  First boot
```
***** Booting Zephyr OS zephyr-v1.14.0-835-g875319b6f646 *****
[00:00:00.004,364] <inf> mcuboot: Starting bootloader
[00:00:00.010,620] <inf> mcuboot: Primary image: magic=unset, copy_done=0x3, image_ok=0x3
[00:00:00.019,165] <inf> mcuboot: Scratch: magic=unset, copy_done=0x0, image_ok=0x3
[00:00:00.027,221] <inf> mcuboot: Boot source: primary slot
[00:00:00.036,102] <inf> mcuboot: Swap type: none
[00:00:00.695,220] <inf> mcuboot: Bootloader chainload address offset: 0xc000
[00:00:00.702,789] <inf> mcuboot: Jumping to the first image slot
***** Booting Zephyr OS zephyr-v1.14.0-837-g703e95a6875e *****
```

- Image upload, test
```
Images:
 slot=0
    version: 1.0.0
    bootable: true
    flags: active confirmed
    hash: c38f76868519461f8a72b908a97be84a8d4e3c3872067d0ae2b506d27a41aa49
 slot=1
    version: 1.0.1
    bootable: true
    flags: pending
    hash: 1e729cfdba73fb8af0cfd2a781512c9d3b61d3dd6d3214d5025a09d7be4b4a56
Split status: N/A (0)
```

1. Reset. Notice the Watchdog reset during swap
```
***** Booting Zephyr OS zephyr-v1.14.0-835-g875319b6f646 *****
[00:00:00.005,432] <inf> mcuboot: Starting bootloader
[00:00:00.011,718] <inf> mcuboot: Primary image: magic=unset, copy_done=0x3, image_ok=0x3
[00:00:00.020,355] <inf> mcuboot: Scratch: magic=unset, copy_done=0x0, image_ok=0x3
[00:00:00.028,442] <inf> mcuboot: Boot source: primary slot
[00:00:00.037,414] <inf> mcuboot: Swap type: test
�***** Booting Zephyr OS zephyr-v1.14.0-835-g875319b6f646 *****
[00:00:00.004,364] <inf> mcuboot: Starting bootloader
[00:00:00.010,620] <inf> mcuboot: Primary image: magic=good, copy_done=0x3, image_ok=0x3
[00:00:00.019,104] <inf> mcuboot: Scratch: magic=unset, copy_done=0x0, image_ok=0x3
[00:00:00.027,130] <inf> mcuboot: Boot source: primary slot
[00:00:08.585,418] <inf> mcuboot: Swap type: none
[00:00:09.266,876] <inf> mcuboot: Bootloader chainload address offset: 0xc000
[00:00:09.274,444] <inf> mcuboot: Jumping to the first image slot
***** Booting Zephyr OS zephyr-v1.14.0-837-g703e95a6875e *****
```
The slot states are:
```
Images:
 slot=0
    version: 1.0.1
    bootable: true
    flags: active
    hash: de40bff8337c57cb756aa50687aaff9cf487fee4c8124b525bd63f3cd4f8854a
 slot=1
    version: 1.0.0
    bootable: true
    flags: confirmed
    hash: c38f76868519461f8a72b908a97be84a8d4e3c3872067d0ae2b506d27a41aa49
Split status: N/A (0)
```
2. reset
```
***** Booting Zephyr OS zephyr-v1.14.0-835-g875319b6f646 *****
[00:00:00.005,432] <inf> mcuboot: Starting bootloader
[00:00:00.011,749] <inf> mcuboot: Primary image: magic=good, copy_done=0x1, image_ok=0x3
[00:00:00.020,294] <inf> mcuboot: Scratch: magic=unset, copy_done=0x0, image_ok=0x3
[... 68 lines cut ...]
[00:00:00.010,620] <inf> mcuboot: Primary image: magic=good, copy_done=0x3, image_ok=0x3
[00:00:00.019,104] <inf> mcuboot: Scratch: magic=unset, copy_done=0x0, image_ok=0x3
[00:00:00.027,160] <inf> mcuboot: Boot source: primary slot
[00:00:08.584,838] <inf> mcuboot: Swap type: none
[00:00:09.245,819] <inf> mcuboot: Bootloader chainload address offset: 0xc000
[00:00:09.253,387] <inf> mcuboot: Jumping to the first image slot
***** Booting Zephyr OS zephyr-v1.14.0-837-g703e95a6875e *****
```
```
Images:
 slot=0
    version: 1.0.0
    bootable: true
    flags: active
    hash: c38f76868519461f8a72b908a97be84a8d4e3c3872067d0ae2b506d27a41aa49
 slot=1
    version: 1.0.1
    bootable: true
    flags: confirmed
    hash: de40bff8337c57cb756aa50687aaff9cf487fee4c8124b525bd63f3cd4f8854a
Split status: N/A (0)
```
...

[binaries_with_dbg_logs.zip](https://github.com/JuulLabs-OSS/mcuboot/files/3190593/binaries_with_dbg_logs.zip)

ccollins476ad, 2019-05-19:
I see this issue as well.  Here is an example of what I see:

```
1. Initial state

0  1.1.1.1: 9195333e13ee6565a5583d6835a24a68be9d937871826c83f26a1cacffc71822 abc
1  2.2.2.2: 63d616db6697fb09c0c4b5d0755c15c23388a429a5e149bfc51cafcf2d4aafdf  b

2. Test image (second slot now pending)

0  1.1.1.1: 9195333e13ee6565a5583d6835a24a68be9d937871826c83f26a1cacffc71822 abc
1  2.2.2.2: 63d616db6697fb09c0c4b5d0755c15c23388a429a5e149bfc51cafcf2d4aafdf  b p

3. Reboot (first slot temporarily active)

0  2.2.2.2: 63d616db6697fb09c0c4b5d0755c15c23388a429a5e149bfc51cafcf2d4aafdf ab
1  1.1.1.1: 9195333e13ee6565a5583d6835a24a68be9d937871826c83f26a1cacffc71822  bc

4. Reboot again (trigger revert operation); interrupt revert with a crash; allow revert to resume and complete.

0  1.1.1.1: 9195333e13ee6565a5583d6835a24a68be9d937871826c83f26a1cacffc71822 ab
1  2.2.2.2: 63d616db6697fb09c0c4b5d0755c15c23388a429a5e149bfc51cafcf2d4aafdf  bc

5. Reboot again (allow extraneous swap to occur)

0  2.2.2.2: 63d616db6697fb09c0c4b5d0755c15c23388a429a5e149bfc51cafcf2d4aafdf abc
1  1.1.1.1: 9195333e13ee6565a5583d6835a24a68be9d937871826c83f26a1cacffc71822  b

The bug occurs in step 4. Specifically, the "c" (confirmed) flag should be set on slot 0, not slot 1. Because of the misplaced confirm flag:

    the subsequent reboot causes an additional swap.
    no additional DFU operations can be performed until the next reboot.
```

I see one additional swap after the revert completes, but then the system settles.  Note: the system only crashed once in my test.  If it crashed on every boot, then I believe I would see continuous swaps.

### Problem

Here is what I believe is happening:

After the reboot in step 4 (reboot from "test" state), mcuboot sees the following image trailers:

|                 | primary slot | secondary slot |
| --------------- | ------------ | -------------- |
|       **magic** | Good         | Unset          |
|    **image-ok** | Unset        | Unset          |
|   **copy-done** | Set          | Unset          |

This correctly translates to the "revert" state.  Mcuboot proceeds to swap the two image slots.  When it swaps the final sector in each slot (this is actually the first swap that occurs), it performs the following procedure:

1. Erase the trailer in the primary slot [^1].
2. Re-initialize the trailer in the primary slot.

In step 2, the primary trailer gets initialized with the following contents:

|                 | primary slot | secondary slot |
| --------------- | ------------ | -------------- |
|       **magic** | Good         | Unset          |
|    **image-ok** | Unset        | Unset          |
|   **copy-done** | Unset        | Unset          |

In other words, the `copy-done` flag gets cleared.  These trailer contents do not map to any swap state, resulting in a `BOOT_SWAP_TYPE_NONE` swap type in the code.  Because an interrupted swap is being resumed, the `BOOT_SWAP_TYPE_NONE` then get converted into `BOOT_SWAP_TYPE_PERM` (in `boot_previous_swap_type()`).  Mcuboot completes the swap and sets the `copy-done` flag in the primary trailer, resulting in the following swap state:

|                 | primary slot | secondary slot |
| --------------- | ------------ | -------------- |
|       **magic** | Good         | Unset          |
|    **image-ok** | Unset        | Unset          |
|   **copy-done** | Set          | Unset          |

So we are back in the "revert" state.  The images get swapped again on the next reboot.

### Solution

I propose we solve this as follows.  If and only if an interrupted swap is being resumed, then mcuboot should interpet the following swap state as "revert":

|                 | primary slot | secondary slot |
| --------------- | ------------ | -------------- |
|       **magic** | Good         | Unset          |
|    **image-ok** | Unset        | Unset          |
|   **copy-done** | Unset        | Unset          |

I implemented a much more complicated solution that seems to work, but upon reflection I think the simpler approach descbired above is sufficient.  I am intereted to hear others' thoughts.

[^1]: We erase the primary trailer and then immediately re-write it.  This seems like it could be problematic in the (very unlikely) case where the system reboot after the erase step.  In this case, the primary trailer would not be present at all.

PR #485 (2019-05-31): Fix double swap on interrupted revert
This fixes #480.

When mcuboot rewrites image trailers during a swap, some information is lost.  If a reset occurs before the swap completes, mcuboot may not be able to determine which swap type to resume upon startup.  Specifically, if a "revert" swap gets interupted, mcuboot will perform an extraneous swap on the subsequent boot.  See https://github.com/JuulLabs-OSS/mcuboot/issues/480 for details.

**NOTE:** A different PR (#482) adds unit tests which test this fix.

This PR adds an additional field to the image trailer: `swap-type`.  The new trailer str [... cut]
commit a1c1204f (2019-05-31): Fix double swap on interrupted revert

This fixes #480.

When mcuboot rewrites image trailers during a swap, some information is
lost.  If a reset occurs before the swap completes, mcuboot may not be
able to determine what which swap type to resume upon startup.
Specifically, if a "revert" swap gets interupted, mcuboot will perform
an extraneous swap on the subsequent boot.  See
https://github.com/JuulLabs-OSS/mcuboot/issues/480 for details.

This commit adds an additional field to the image trailer: `swap-type`.
The new trailer structure is illustrated below:

```
     0                   1   [... cut]
closed 2019-05-31 by ccollins476ad