mcu-tools/mcuboot #486: imgtool supports verifying signed firmware binaries

robbawebba, 2019-05-27:
I recently asked on the mcuboot slack channel if it's possible to verify if a firmware binary is signed properly, whether it be with imgtool or even openssl. It turns out this feature is not implemented but might be useful in the future.

The main benefit of this feature that I could think of, besides checking if the binary is signed, would be to determine which private key was used to sign a given firmware image. In development I use the default key provided by this repository, but I use a separate key for deployment in production. I've run into issues in the past where there was a mismatch in signing keys, so this feature would be very helpful for debugging this issue if it occurs, as well as validating a signed image before deploying it.

I imagine this feature would be another command built into imgtool.py. The user would have to provide a path to the signed firmware image and the path to the private key for verification. I'm not very familiar with the way images are signed, so I would appreciate anyone's input on this feature.

utzig, 2019-05-29:
Support for verification was added with #487. Still lacks support for encrypted images but that should arrive eventually or another issue can be created for it.

closed 2019-05-29 by utzig

robbawebba, 2019-05-29:
@utzig Thank you for implementing this feature! My team and I really appreciate it :100: