mcu-tools/mcuboot #489: Insufficient image encryption feature

PhilippFinke, 2019-06-04:
I want to encrypt my image to prevent potential attackers to analyse it. As far as I understood the encryption feature works as follows:

1. The image will be encrypted after build using a randomly generated key.
2. The randomly generated key will be encrypted with a fixed key defined by mcuboot (defined in boot/zephyr/keys.c).
3. The encrypted randomly generated key will be attached to the FW image.

This will lead to following security leak:

1. An attacker can read/sniff the encrypted image.
2. The attacker uses the openly accessible key from keys.c to decrypt the randomly generated key contained in the image.
3. The attacker uses the decrypted randomly generated key to decrypt the complete image.

Maybe I'm wrong here, but if this is true the encryption feature doesn't increase security at all and is completely useless.

Hope you can give me feedback to this issue.

utzig, 2019-06-04:
Yes, it is true if you are not able to secure your private key. `keys.c` contains just samples for people to use, you should provide a safe way to store your key and access it; if you can't secure your keys in your HW, no method will ever provide you with security.

PhilippFinke, 2019-06-04:
Why isn't there a way to generate an encryption key during build similar to the signing key method, where I'm able to provide a keyfile during build? That somehow doesn't make sense to me.

utzig, 2019-06-04:
Public key encryption requires the private key to decrypt; if it used the public key anyone would be able to decrypt, no?

PhilippFinke, 2019-06-04:
That's true. But wouldn't it be nice to provide a private key to the build of mcuboot that I randomly created on my machine?

utzig, 2019-06-04:
Not sure I understand what you mean; you can totally create a random keypair and add the private part to `mcuboot`, while keeping the public part available to encrypt.

PhilippFinke, 2019-06-04:
In keys.c the private key used for image decryption is saved in enc_priv_key[]. This key is openly accessible by everybody. This allows the easy encryption of the FW image.

Is there any possibility to exchange this hardcoded key to one I created?

Laczen, 2019-06-04:
@utzig, what @PhilippFinke means is: why is there no method like getpub for getting private keys to include in keys.c
@PhilippFinke what I think you need to to is extend general.py under imgtool/keys to provide a export functionality for private keys, and to extend imgtool.py with a getpriv option.

PhilippFinke, 2019-06-04:
Is there anybody who can solve this faster and better than me? I'm not really into mcuboot and definitely a zero in Python.

d3zd3z, 2019-11-27:
Does someone want to make this patch into a pull request? Otherwise, we should probably close this issue as resolve.

gon1332, 2019-11-28:
@d3zd3z I look into it.

closed 2021-06-03 by github-actions[bot]