mcu-tools/mcuboot #493: How as a user can confirm that the image is OK?

gon1332, 2019-06-20:
After reading the docs and running the bootloader, I understood that there is also a user method of marking an image as OK (image trailer specific field). After that, either a permanent swap will be performed, or a revert.

Is there an API I can refer into?

utzig, 2019-06-20:
You can call `boot_set_confirmed` from your application; it will write `image_ok` to the primary slot:

https://github.com/JuulLabs-OSS/mcuboot/blob/master/boot/bootutil/src/bootutil_misc.c#L684

gon1332, 2019-06-21:
Moreover for the Zephyr users, the wrapper is [`boot_write_img_confirmed`](https://github.com/zephyrproject-rtos/zephyr/blob/master/include/dfu/mcuboot.h#L145).

closed 2019-09-09 by utzig