mcu-tools/mcuboot #494: Trailer size and its impact on boot_swap_sectors

solsbarry, 2019-06-25:
Hi,
I am new to MCUboot, so forgive me if I am off with this.

In loader.c, the function boot_swap_sectors, when determining whether to "use_scratch" you want to make sure that the image is not so big that it would occupy the area where the trailer goes. There is an assumption here that the trailer only needs one sector (the last sector).  thus the "-1" in this line :
`last_sector = boot_img_num_sectors(&boot_data, BOOT_PRIMARY_SLOT) - 1;`
But If your slots are big enough and you can make BOOT_STATUS_MAX_ENTRIES big enough, and maybe your min write size is 4,  and your sectors are small enough, then your trailer could take up 2 or more sectors. 
So then I would think you might want to calculate how many sectors your trailer takes up, and subtract that from the last sector index. Something like this:
```
sector_sz = (*boot_data.imgs[BOOT_PRIMARY_SLOT].sectors).fs_size;
num_trailer_sectors = trailer_sz / sector_sz  + ( trailer_sz % sector_sz > 0 ? 1 : 0);
last_sector = boot_img_num_sectors(&boot_data, BOOT_PRIMARY_SLOT) - num_trailer_sectors;
```
This compiles, although I haven't actually tried the code I wrote above. I don't actually have this issue now, but I think I will eventually.

utzig, 2019-06-25:
Nice one, this might be a problem, but most probably not. While the trailer may span more than one sector, while we are swapping the last sectors, only one status entry was used which should only amount to a tens of bytes. The trailer is big because it has to store status for all sectors swapped, but by the point where we are writing on the `last - 1` sector we already swapped much of the sectors. This could be a problem though if sectors are very small, but smaller I've ever seen is 256 bytes.

solsbarry, 2019-06-26:
That makes sense. I hadn't thought about the fact that by the time we would be writing over the trailer, the only relevant info in the trailer is right at the end.

closed 2019-06-26 by solsbarry